LibrePCB/LibrePCB #989: Installed as snap on Fedora - but core dumps

tinpotnick, 2022-07-06:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.6
Git Revision: c4f316428
Qt Version: 5.9.5
Operating System: Fedora release 36
...
```

##### SUMMARY

After installing snap: `snap install librepcb` then running via librepcb it segment faults with the output:

Qt: Session management error: None of the authentication protocols specified are supported
Load stroke font: "newstroke.bene"
Start loading font "/snap/librepcb/621/usr/local/share/librepcb/fontobene/newstroke.bene"
[  INFO   ] "LibrePCB 0.1.6 (c4f316428)" (:0)
[  INFO   ] "Qt version: 5.9.5 (compiled against 5.9.5)" (:0)
[  INFO   ] "Resources directory: /snap/librepcb/621/usr/local/share/librepcb" (:0)
[  INFO   ] "Application settings: /home/nick/snap/librepcb/621/.config/LibrePCB/LibrePCB.ini" (:0)
[DEBUG-MSG] Started network access manager thread. (:0)
[DEBUG-MSG] Recently used workspace: "" (:0)
Fontconfig warning: FcPattern object weight does not accept value [0 205)
Segmentation fault (core dumped)

##### STEPS TO REPRODUCE

Simply run librepcb on fedora 36

##### EXPECTED RESULTS

The application.

##### ACTUAL RESULTS

Core dump

##### ADDITIONAL INFORMATION

(gdb) bt
#0  0x00007faa8d8aaa04 in ?? () from /snap/librepcb/621/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#1  0x00007faa941ccc71 in ?? () from /snap/librepcb/621/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#2  0x00007faa941d02ed in QFontDatabase::findFont(QFontDef const&, int) () from /snap/librepcb/621/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#3  0x00007faa941d0b16 in QFontDatabase::load(QFontPrivate const*, int) () from /snap/librepcb/621/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#4  0x00007faa941a6543 in QFontPrivate::engineForScript(int) const () from /snap/librepcb/621/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#5  0x00007faa941c7f51 in QFontMetricsF::leading() const () from /snap/librepcb/621/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#6  0x00007faa943723c8 in ?? () from /snap/librepcb/621/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#7  0x00007faa94372f30 in ?? () from /snap/librepcb/621/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#8  0x00007faa941c7683 in QFontMetrics::boundingRect(QRect const&, int, QString const&, int, int*) const () from /snap/librepcb/621/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#9  0x00007faa941c76fa in QFontMetrics::size(int, QString const&, int, int*) const () from /snap/librepcb/621/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#10 0x00007faa94a43ca4 in QPushButton::sizeHint() const () from /snap/librepcb/621/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x00007faa948d5b14 in QWidgetItemV2::updateCacheIfNecessary() const () from /snap/librepcb/621/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#12 0x00007faa948d5dfd in QWidgetItemV2::maximumSize() const () from /snap/librepcb/621/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#13 0x00007faa948ca422 in ?? () from /snap/librepcb/621/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#14 0x00007faa948cb0f6 in QBoxLayout::minimumSize() const () from /snap/librepcb/621/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#15 0x00007faa948ce3d6 in ?? () from /snap/librepcb/621/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007faa948ceab5 in ?? () from /snap/librepcb/621/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#17 0x00007faa948ced0d in QGridLayout::minimumSize() const () from /snap/librepcb/621/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#18 0x00007faa948d1bcd in QLayout::totalMinimumSize() const () from /snap/librepcb/621/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#19 0x00007faa94ad9f74 in ?? () from /snap/librepcb/621/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#20 0x00007faa94addf8a in ?? () from /snap/librepcb/621/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#21 0x00007faa94adeb95 in QWizard::setWizardStyle(QWizard::WizardStyle) () from /snap/librepcb/621/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#22 0x0000558a9d4b5a00 in librepcb::application::FirstRunWizard::FirstRunWizard(QWidget*) ()
#23 0x0000558a9d4a92a3 in main ()

ubruhin, 2022-07-30:
Thanks for the bugreport. Have you already tried to install LibrePCB in a different way (e.g. the AppImage or the portable binary package from our website) to see if generally it would run on your system, i.e. there must be something wrong with the snap package?

andsmedeiros, 2023-08-02:
I also experience this:

```
AAA@BBB:~$ librepcb
Qt: Session management error: Could not open network socket
Found stroke font: "newstroke.bene"
Start loading stroke font  "/snap/librepcb/886/usr/local/share/librepcb/fontobene/newstroke.bene" in worker thread...
[  INFO   ] LibrePCB 0.1.7 (ba553c6fb)
[  INFO   ] Qt version: 5.9.5 (compiled against 5.9.5)
[  INFO   ] Resources directory: "/snap/librepcb/886/usr/local/share/librepcb"
[  INFO   ] Application settings: "[HOME REDACTED]/snap/librepcb/886/.config/LibrePCB/LibrePCB.ini"
[DEBUG-MSG] Network access manager thread started.
[ WARNING ] propsReply "An AppArmor policy prevents this sender from sending this message to this recipient; type=\"method_call\", sender=\":1.11711\" (uid=1000 pid=1772327 comm=\"/snap/librepcb/886/usr/local/bin/librepcb\" label=\"snap.librepcb.librepcb (enforce)\") interface=\"org.freedesktop.DBus.Properties\" member=\"GetAll\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.freedesktop.NetworkManager\" (uid=0 pid=2338 comm=\"/usr/sbin/NetworkManager --no-daemon\" label=\"unconfined\")"
[ WARNING ] nmReply "An AppArmor policy prevents this sender from sending this message to this recipient; type=\"method_call\", sender=\":1.11711\" (uid=1000 pid=1772327 comm=\"/snap/librepcb/886/usr/local/bin/librepcb\" label=\"snap.librepcb.librepcb (enforce)\") interface=\"org.freedesktop.NetworkManager\" member=\"GetDevices\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.freedesktop.NetworkManager\" (uid=0 pid=2338 comm=\"/usr/sbin/NetworkManager --no-daemon\" label=\"unconfined\")"
[ WARNING ] "Object path cannot be empty"
[DEBUG-MSG] Recently used workspace: ""
Fontconfig warning: FcPattern object weight does not accept value [0 205)
Segmentation fault (core dumped)
```

I downloaded the AppImage and it loads normally, so it must be the snap packaging.

ubruhin, 2023-08-03:
Hmm although there are some unusual warnings in the console output, it's not clear whether they are related to the segfault or not. But somehow it could indeed be related to fonts, as the stacktrace from the initial post ends up in some Qt font function.

Maybe the workaround described in https://github.com/keshavbhatt/olivia/issues/95#issuecomment-774747492 works for LibrePCB too?

andsmedeiros, 2023-08-03:
@ubruhin That solved it, thanks!

ubruhin, 2023-08-03:
Awesome, thanks for testing @andsmedeiros!

So then the problem seems to be somehow related to Snap and/or Qt, nothing we can fix on our side. Maybe we should mention this workaround in our installation instructions :thinking:

ubruhin, 2023-08-03:
The workaround is now mentioned in the installation instructions: https://librepcb.org/docs/installation/linux/#snap

I think there's nothing else we can do regarding this problem, so I'll close the issue.

closed 2023-08-03 by ubruhin