LibrePCB/LibrePCB #990: text not updated during "Update Library"

keredson, 2022-07-27:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 0.1.6
Git revision: c4f316428
Build date: 2021-11-03 21:59:38 (PDT)
Operating System: Ubuntu 22.04 LTS
```

##### SUMMARY
Board Editor does not reflect text changes from the Library Editor during "Update Library".  You have to delete and re-place the device.

##### STEPS TO REPRODUCE

1. place a device in the Board Editor
2. add text to device's component in the Library Editor; save
3. run "Update Library" in Board Editor
4. text does not appear in the already placed device
5. delete the device and re-place it and the text will appear

##### EXPECTED RESULTS
Text will update.

##### ACTUAL RESULTS
Text does not update.

ubruhin, 2022-07-30:
This is expected behavior to avoid loosing manual modifications of texts. Maybe the library updater should provide updating the texts as an optional feature :thinking: 

> You have to delete and re-place the device.

Right-click on the device and choosing "Reset all texts" should do the same but without loosing the traces.

AustraleanPerson, 2026-10-10:
As an alternative solution, I can suggest to avoid adding text labels to `librepcb_schematic->symbol` if they have not been overridden in symbol instance. If text is deleted in symbol instance, then add something like this:
`(text (original 3c37a956-c23e-419c-81c8-f8fdf422bb8f) (value none))`
And no problems with update. This will also help avoid redundant/duplicate information.

ubruhin, 2026-10-10:
Agree to @AustraleanPerson, probably that is the best solution (details to be defined how to implement it exactly).