LibrePCB/LibrePCB #1001: Add "assembly type" property to packages & device instances

ubruhin, 2022-09-08:
##### SUMMARY

For automated assembly, we need to know the assembly type of each device on the board, i.e. whether it's a THT or SMT package. In particular, this information might be printed in pick&place files to improve automated processing & quotation for the assembly process.

In addition, this property could also be used to mark packages/devices which are not real parts - for example logos on the silkscreen, fiducials etc. Since we don't have this property yet, the BOM/pick&place output even contains devices like "LibrePCB Logo" which is very ugly.

Note that Gerber X3 supports the following values for the `.CMnt` (mount type) property : `TH`, `SMD`, `Fiducial`, `Other`

##### SOLUTION

Some thoughts:

- Probably we need this property on library packages (not on packages, and not on footprints).
- To ensure this property is always set to a sane value (not just left at a possibly wrong default value):
  - either it should be asked in the package creation wizard
  - or it should be set to some kind of "auto" value, which will detect SMT or THT by the available footprint pads
  - or it should be set to some kind of "unset" value, which causes a library check warning asking the user to specify a proper value
- Within a board, each device instance probably should allow to (optionally) override this property with a different value. By default, the value from the library shall be used.
- Probably we need at least these values: THT, SMT, Fiducial
- Maybe some more values would be helpful as well: Mixed (THT/SMT), Mechanical, No Assembly
- Decide how it should be named, "assembly type", "mount type", "package type", ...?

Of course the BOM- and pick&place generators shall respect this property to decide whether to include or exclude a device.

EdizonTN, 2023-03-11:
Perfect! This will be very helpfull.

> Of course the BOM- and pick&place generators shall respect this property to decide whether to include or exclude a device.

Also assembly printouts have to accept these settings! Very important for us! Please count with this.

ubruhin, 2023-03-11:
> Also assembly printouts have to accept these settings! Very important for us! Please count with this.

Hm, either I interpret your comment wrongly or you misunderstand this feature :see_no_evil: This feature is not mainly about which device gets assembled or not, but just what kind of assembly a package needs - i.e. *how* a device needs to be assembled (SMT or THT), not *if*. This property is stored in the library, not in the board.

However, assembly variants are planned for 0.2.0 as well, see #965 :slightly_smiling_face: For that, it makes sense to reflect the assembly variant in printouts :+1:

closed 2023-03-13 by ubruhin