LibrePCB/LibrePCB #1028: Add shortcut to remove last polygon/plane/trace segment

ubruhin, 2022-09-21:
##### SUMMARY

Sometimes one places a polygon or plane vertex or a trace segment at the wrong position. Then you have to abort the whole tool, delete the segment, and restart the tool. It would be convenient to have a shortcut for exactly this situation.

##### SOLUTION

Maybe the 'Backspace' key would be intuitive for that. It should just remove the last positioned line segment without leaving the tool. This should be available in all polygon-, plane- and trace tools in all editors. The shortcut should be configurable of course.

dbrgn, 2022-09-24:
Other ideas for a shortcut: "u" (undo) or "ctrl+z"?

doom-fr, 2023-08-27:
linked to https://github.com/LibrePCB/LibrePCB/issues/606