LibrePCB/LibrePCB #1039: Optionally initialize new schematic with a schematic frame

ubruhin, 2022-09-27:
##### SUMMARY

Currently a newly created schematic is completely empty. Especially for beginners it's not obvious that you should add a schematic frame with the "Add Component" dialog first to get the typical schematic look. Therefore it would be convenient if LibrePCB suggests to automatically initialize a new schematic with a frame. Even for experienced users this would be useful since it allows to safe time.

##### SOLUTION

I'm not sure yet. Somehow LibrePCB needs to know which of the components in the library are schematic frames, and somehow it needs to know a sane default value (probably A4 landscape). But how should it know that? I don't like to hardcode some UUIDs since this wouldn't work with custom libraries.

Anyway, probably it would be nice to allow the user to override the default schematic frame, which should then be stored in the workspace settings. And if a project already contains at least one schematic, maybe the frame from an existing schematic should automatically be applied to newly created schematics (instead of using the workspace setting)?

Tubbles, 2023-02-13:
Maybe have all the required settings for this in the workspace settings? Like an elaborate uuid list picker of all the component groups that counts as being a group of "frames" and then also a a setting for a specific component uuid which is the default frame

AvetosDesign, 2026-09-16:
Other CAD packages treat this type of thing with "templates". They're typically just a design file that is used to jump-start a project.  More often than not, they are constructed using the same drawing tools that the CAD designs use, but the template file itself is treated as read-only.

So one way it could work here would be to create a few templates (A4 Landscape, A4 Portrait, ...) by placing the appropriate frames and saving the files.  You then add a "Default template" to the user settings.  Any new projects then automatically use that template as a starting point.  You could extend the Project settings to support replacing the template if so desired.