LibrePCB/LibrePCB #1050: Remove Close Button?

dbrgn, 2022-10-02:
##### SUMMARY

The "Close Project" shortcut button is not something I use often (or ever, actually). At the same time it's easy to accidentally hit when wanting to click on the "File" menu (just happened to me).

[image]

##### SOLUTION

If the user wants to close the project, there are two other ways:

- Menu: File > Close Project
- Keyboard: Ctrl+F4

I think these should be sufficient, and the toolbar space should not be wasted with a button that causes annoyance when hit accidentally.

ubruhin, 2022-10-02:
Actually I use this button very often :see_no_evil:

Maybe it would become obsolete automatically when implementing #53...

dbrgn, 2022-10-02:
Ok, then I guess our habits differ 😄 I rarely think about "project closing", more about "window closing". I also close the schematics editor window when working on the board editor window (and vice versa), because the windows bother me. So #53 might indeed be an improvement, because then "closing the window" would equal "closing the project", if I understand the proposal correctly.

closed 2025-05-13 by ubruhin