LibrePCB/LibrePCB #1052: Support Qt6

ubruhin, 2022-10-02:
##### SUMMARY

It should be clear that we should add support for Qt6 for long term, so let's open this issue to keep it in mind :slightly_smiling_face: 

##### SOLUTION

- Support building with Qt6, while still keeping compatibility with Qt5
- Build & deploy all official binary releases with Qt6

ubruhin, 2023-12-14:
Actually I wonder if we really need to keep support for Qt5 when migrating to Qt6. Maybe Qt6.2 would be a reasonable minimum requirement. As far as I can see, the consequences of that would be:

### Windows

Qt6 does not support Windows 7/8/8.1 anymore, so Windows 10 or later is required. Generally I really don't like that (market share of Windows 7 is still around 3%), but all Windows versions older than 10 are EOL anyway. In addition, even after releasing LibrePCB 1.1.0, users on Windows 7 can still continue to use LibrePCB 1.0.0 since the file format is still the same. So they still have plenty of time to switch to a newer Windows version.

### macOS

According [Qt 6.2 requirements](https://doc.qt.io/qt-6.2/supported-platforms.html), it should still support macOS 10.14. Even our official macOS releases do not run on 10.14 anymore, so I guess this should be fine.

### Linux

Qt6 is not available on the still supported Ubuntu 20.04. But according [Qt 6.2 requirements](https://doc.qt.io/qt-6.2/supported-platforms.html), it should still be compatible with Ubuntu 20.04 so probably our official binary releases should continue to work on Ubuntu 20.04. Ubuntu 18.04 is EOL already, so it shouldn't matter if LibrePCB doesn't run anymore there. I suspect for Debian is should be similar.

Unfortunately Qt 6.2 is still relatively young (~2 years), maybe it is not available yet in some Linux distributions(?). This would be a problem to create LibrePCB packages for those distributions. Also we need to check if Flatpak and Snap already provide Qt6 flawlessly.

Anyway, I think we should check if supporting Qt5 and Qt6 at the same time is a big pain. If it works fine, it might be better to provide a transition period where LibrePCB works with both versions (a hard cut is always risky and cumbersome for package maintainers).

dbrgn, 2023-12-14:
Having a single release that supports both versions (with big deprecation warnings in the release notes) might be nice. But after that, I think requiring Qt 6 (i.e. dropping support for Qt 5) should be fine in my view.

You can check for Qt6 availability here: https://pkgs.org/search/?q=qt6 (make sure to look at the actual package names, the names aren't the same across all distros.)

ubruhin, 2023-12-15:
> Having a single release that supports both versions (with big deprecation warnings in the release notes) might be nice. But after that, I think requiring Qt 6 (i.e. dropping support for Qt 5) should be fine in my view.

Sounds like a plan :+1:

> You can check for Qt6 availability here: https://pkgs.org/search/?q=qt6 (make sure to look at the actual package names, the names aren't the same across all distros.)

Ah, nice page, but a bit cumbersome to search for Qt since many many application packages have "qt6" in their name :see_no_evil: But as far as I can see, the spread of Qt6 looks not bad.

nimish, 2023-12-26:
> ### macOS
> According [Qt 6.2 requirements](https://doc.qt.io/qt-6.2/supported-platforms.html), it should still support macOS 10.14. Even our official macOS releases do not run on 10.14 anymore, so I guess this should be fine.
>

Apple only supports the last 3 versions of macOS. Why carry the baggage?

ubruhin, 2023-12-27:
> Why carry the baggage?

Hm, which baggage?

nimish, 2023-12-27:
> > Why carry the baggage?
> 
> Hm, which baggage?

The maintenance burden of supporting obsolete macOS versions.

ubruhin, 2023-12-27:
> > > Why carry the baggage?
> > 
> > 
> > Hm, which baggage?
> 
> The maintenance burden of supporting obsolete macOS versions.

I'm a bit confused, maybe you misread the sentence you quoted? As far as I can see, I never wrote we want to support obsolete macOS versions, I only wrote what macOS versions are supported by Qt 6.2 :see_no_evil: Btw, actually we even got bug reports from macOS users after LibrePCB 1.0 has dropped support for macOS 10.14 :wink:

nimish, 2023-12-27:
Ah, I see -- no worries then.

closed 2024-03-22 by ubruhin