LibrePCB/LibrePCB #1054: Schematics: Add checks/warnings for unconnected wires

dbrgn, 2022-10-07:
##### SUMMARY

On a recent project, I debugged a non-working board on the workbench for half an hour... Until I found this:

[image]

The connection wasn't done properly, yet no ERC warning was emitted.

##### SOLUTION

- If a wire is not connected on both sides, show a warning in the ERC about an unconnected wire
- To silence this warning, either connect the wire or explicitly mark it as unconnected
- To handle the case where the wire is only used to place a netlabel: Warn if a netlabel is attached that only exists once in the schematic. Do not warn if a netlabel is attached that is also present on other wires.

Prior art: In KiCAD, connections must be marked as unconnected as well.

[image]

The tool to do so can be found in the toolbar, next to the tools used to draw wires, add labels, etc.

[image]

To me, this solution feels quite natural. We could even add a way to add these no-connection flags through the ERC (right-click warning, choose "Add no-connection flag").

ubruhin, 2022-10-07:
I agree that in your example there should appear an ERC error. Just some comments:

> If a wire is not connected on both sides, show a warning in the ERC about an unconnected wire

Sounds reasonable, I don't see a case where such a warning would be annoying :+1:

> Warn if a netlabel is attached that only exists once in the schematic.

Indirectly this is already the case. If there's a net (no matter of how many split segments it consists of) which is connected to less than two symbol pins, an ERC warning is raised. This covers even more error cases.

> Prior art: In KiCAD, connections must be marked as unconnected as well.

This is one of the (many) reasons why I don't like KiCad ;) IMHO it's stupid if the schematic editor assumes that EVERY pin needs to be connected, since in real world that's (by far) not the case. Therefore in LibrePCB the information of whether a pin is required or optional is specified in the component library element, and the ERC warns only about unconnected *required* pins since this is not annoying but still very helpful. However, it might make sense to optionally allow configuring the ERC to be more restrictive :thinking: 

But, since all these variants still don't cover your specific problem: I see two options for that one:

- Either warn if a line ends on another line without having a connection, i.e. detect the *visual* issue instead of a *logic* issue. This would cover the most tricky cases, but it would still not warn if there's no overlap (e.g. if the line in your example was 1 grid interval shorter).
- Or, warn about lines with open ends which do not have a netlabel assigned, since open ends make sense only for netlabels. However, LibrePCB doesn't know to which line a netlabel belongs to (a netlabel is a floating object), this is a bit inaccurate as well - there could be a netlabel far away from the open end which unintentionally suppresses the ERC warning.

Both variants would cover your specific case. But the first one is much more complicated to implement :see_no_evil:

dbrgn, 2022-10-08:
> This is one of the (many) reasons why I don't like KiCad ;) IMHO it's stupid if the schematic editor assumes that EVERY pin needs to be connected, since in real world that's (by far) not the case.

That's not what I was suggesting, only if a wire is attached to the pin 🙂

This is fine (no wires attached):

[image]

This is not OK (wire attached, no netlabel):

[image]

This is not OK (wire attached, with netlabel, but netlabel is not attached to any other net). I think this already results in a warning.

[image]

This is OK, wire is attached through netlabel.

[image]

Tubbles, 2023-02-13:
Related #823

PR #1185 (2023-08-06): ERC: Warn about schematic wires with open end
As requested in #1054, this adds an ERC check for open wires in the schematic, i.e. wires with an unconnected end and no net label:

[image]

Note that this warning is suppressed if the "Less than two pins in net" warning is raised on the same net segment because these two warnings are very similar and thus annoying to get both of them in many cases. The new warning is only raised if the wire *does* connect multiple pins and has no net label attached.

Fixes #1054
closed 2023-08-06 by ubruhin