LibrePCB/LibrePCB #1055: Device - Copy from sch. and paste to lib. Or - How to reuse device.

EdizonTN, 2022-10-12:
##### SUMMARY
I have a created schematic and PCB. Whole project. But I haven't source libraries.
But How I can a reuse some device from this project (schematic symbol+PCB package+Component+Device+categories) into a new project?
No way right now...

##### SOLUTION
1. In the schematic editor, I select a component and copy them to clipboard
2. I open a Library manager, select some library and open it. 
3. Next I select new command "PASTE" and whole device will be add to library.
Easy, fast, useful. :-)

[image]

ubruhin, 2022-10-12:
Double-click the symbol in the schematic editor. Then click on these two links at the bottom:

[image]

This will open the corresponding project library directories in your file manager. Then copy these directories into your local library. If you also need package and device, do the same in the board editor. Afterwards trigger a workspace library rescan and wait until the library editor shows the new elements. Note that categories are not copied into projects so they cannot be extracted.

But I agree we should provide a simpler command for that :+1:

EdizonTN, 2022-10-12:
Thanks @ubruhin. 
I know way of the copy files to lib folder using filemanager. But it's..... hmmm. you know. :-)
Furthermore, this way don't copy new category of the component or package... 
And I need to copy category also.

ubruhin, 2022-10-12:
>  Furthermore, this way don't copy new category of the component or package...
> And I need to copy category also.

As I wrote above, this is not possible. Categories are not bundled within projects so there's nothing which can be extracted. Categories are only needed at the time when *adding* new components to a project, afterwards they have no purpose anymore so it makes no sense to bundle them in projects.

ubruhin, 2023-10-02:
> To solve this issue, there could be helpful to view "current project library" where all currently used parts in schematic/board could be listed.

Yes this might be the way to go. However, there is a fundamental refactoring of the whole editor UI planned soon, which would make lots of efforts on this topic obsolete. So at the moment we should wait with this feature until the refactoring is finished (or keep it extremely simple for now, as a temporary solution).