LibrePCB/LibrePCB #1057: Once more potencial problem with march not x86_64

ktrace, 2022-10-20:
LibrePCB Version: 0.1.7
Git Revision: ba553c6fb
Qt Version: 5.15.5
Operating System: gentoo

```

##### SUMMARY
With march=native on some CPU fail two tests:

##### STEPS TO REPRODUCE
Build with march not x86_64 and run tests.

##### EXPECTED RESULTS
All tests OK

##### ACTUAL RESULTS
[  FAILED  ] 2 tests, listed below:
[  FAILED  ] BoardGerberExportTest.test
[  FAILED  ] BoardPlaneFragmentsBuilderTest.testFragments

##### ADDITIONAL INFORMATION
Will be attached.

ktrace, 2022-10-20:
/var/tmp/portage/sci-electronics/librepcb-0.1.7/work/librepcb-0.1.7/tests/unittests/core/project/board/boardgerberexporttest.cpp:117: Failure

Expected equality of these values:
  expected.toStdString()
    Which is: "G04 --- HEADER BEGIN --- *\nG04 #@! TF.GenerationSoftware,LibrePCB,LibrePCB,0.1.2*\nG04 #@! TF.CreationDate,2019-01-02T03:04:05*\nG04 #@! TF.ProjectId,test_project,7c04f9d5-7366-4c4b-9e17-852ed57b3966,v1*\nG04 #@! TF.Part,Single*\nG04 #@! TF.SameCoordinates*\nG04 #@! TF.FileFunction,Copper,L2,Bot*\nG04 #@! TF.FilePolarity,Positive*\n%FSLAX66Y66*%\n%MOMM*%\nG01*\nG75*\nG04 --- HEADER END --- *\nG04 --- APERTURE LIST BEGIN --- *\nG04 #@! TA.AperFunction,ComponentPad*\n%AMROTATEDOBROUND10*1,1,1.27,0.110267,-0.625353*1,1,1.27,-0.110267,0.625353*20,1,1.27,0.110267,-0.625353,-0.110267,0.625353,0*%\n%ADD10ROTATEDOBROUND10*%\n%AMROTATEDRECT11*20,1,1.27,-1.27,0.0,1.27,0.0,120.0*%\n%ADD11ROTATEDRECT11*%\nG04 #@! TA.AperFunction,SMDPad,CuDef*\n%AMROTATEDOCTAGON12*4,1,8,1.0,0.207107,0.707107,0.5,-0.707107,0.5,-1.0,0.207107,-1.0,-0.207107,-0.707107,-0.5,0.707107,-0.5,1.0,-0.207107,1.0,0.207107,100.0*%\n%ADD12ROTATEDOCTAGON12*%\n%AMROTATEDRECT13*20,1,1.0,-1.0,0.0,1.0,0.0,120.0*%\n%ADD13ROTATEDRECT13*%\nG04 #@! TA.AperFunction,ComponentPad*\n%AMROTATEDOBROUND14*1,1,1.27,0.3175,-0.549926*1,1,1.27,-0.3175,0.549926*20,1,1.27,0.3175,-0.549926,-0.3175,0.549926,0*%\n%ADD14ROTATEDOBROUND14*%\n%AMROTATEDOCTAGON15*4,1,8,1.5,0.414214,0.914214,1.0,-0.914214,1.0,-1.5,0.414214,-1.5,-0.414214,-0.914214,-1.0,0.914214,-1.0,1.5,-0.414214,1.5,0.414214,50.0*%\n%ADD15ROTATEDOCTAGON15*%\nG04 #@! TA.AperFunction,SMDPad,CuDef*\n%AMROTATEDOBROUND16*1,1,1.0,-0.433013,-0.25*1,1,1.0,0.433013,0.25*20,1,1.0,-0.433013,-0.25,0.433013,0.25,0*%\n%ADD16ROTATEDOBROUND16*%\nG04 #@! TA.AperFunction,Conductor*\n%ADD17C,0.2*%\nG04 #@! TA.AperFunction,NonConductor*\n%ADD18C,0.2*%\nG04 #@! TA.AperFunction,ComponentPad*\n%AMROTATEDOBROUND19*1,1,1.27,-0.596705,-0.217183*1,1,1.27,0.596705,0.217183*20,1,1.27,-0.596705,-0.217183,0.596705,0.217183,0*%\n%ADD19ROTATEDOBROUND19*%\n%ADD20R,2.54X1.27*%\nG04 #@! TA.AperFunction,SMDPad,CuDef*\n%AMROTATEDOCTAGON21*4,1,8,1.0,0.207107,0.707107,0.5,-0.707107,0.5,-1.0,0.207107,-1.0,-0.207107,-0.707107,-0.5,0.707107,-0.5,1.0,-0.207107,1.0,0.207107,20.0*%\n%ADD21ROTATEDOCTAGON21*%\nG04 #@! TA.AperFunction,ComponentPad*\n%ADD22O,2.54X1.27*%\n%AMROTATEDOCTAGON23*4,1,8,1.5,0.414214,0.914214,1.0,-0.914214,1.0,-1.5,0.414214,-1.5,-0.414214,-0.914214,-1.0,0.914214,-1.0,1.5,-0.414214,1.5,0.414214,70.0*%\n%ADD23ROTATEDOCTAGON23*%\nG04 #@! TA.AperFunction,Conductor*\n%ADD24C,0.0*%\n%ADD25C,0.01*%\nG04 #@! TA.AperFunction,ViaPad*\n%ADD26R,2.0X2.0*%\n%AMROTATEDOCTAGON27*4,1,8,1.0,0.414214,0.414214,1.0,-0.414214,1.0,-1.0,0.414214,-1.0,-0.414214,-0.414214,-1.0,0.414214,-1.0,1.0,-0.414214,1.0,0.414214,0.0*%\n%ADD27ROTATEDOCTAGON27*%\nG04 #@! TA.AperFunction,Conductor*\n%ADD28C,0.5*%\nG04 #@! TA.AperFunction,ViaPad*\n%ADD29C,2.0*%\nG04 #@! TD*\nG04 --- APERTURE LIST END --- *\nG04 --- BOARD BEGIN --- *\nD10*\nG04 #@! TO.N,N/C*\nG04 #@! TO.C,D2*\nG04 #@! TO.P,D2,4,4*\nX60012659Y41970886D03*\nD11*\nG04 #@! TO.N,N3*\nG04 #@! TO.P,D2,2,2*\nX55613250Y39430886D03*\nD12*\nG04 #@! TO.N,Hello*\nG04 #@! TO.P,D2,7,6*\nX62892955Y31902068D03*\nD13*\nG04 #@! TO.N,N/C*\nG04 #@! TO.P,D2,5,5*\nX58493546Y29362068D03*\nD14*\nG04 #@! TO.N,N3*\nG04 #@! TO.P,D2,1,1*\nX53413546Y38160886D03*\nD15*\nG04 #@! TO.N,GND*\nG04 #@! TO.P,D2,3,3*\nX57812955Y40700886D03*\nD16*\nG04 #@! TO.N,Hello*\nG04 #@! TO.P,D2,6,6*\nX60693250Y30632068D03*\nD17*\nG04 #@! TD.P*\nG04 #@! TO.N,*\nX54683546Y35961182D02*\nX57223546Y31561773D01*\nX58153250Y35031477D01*\nX54683546Y35961182D01*\nD18*\nX63131520Y35615525D02*\nX63298186Y35326850D01*\nX62844967Y35065184D02*\nX63710992Y35565184D01*\nX63472658Y35977990D01*\nX62311633Y35988945D02*\nX62400232Y35931043D01*\nX62465498Y35913555D01*\nX62571178Y35919400D01*\nX62819439Y36062733D01*\nX62877341Y36151332D01*\nX62894829Y36216598D01*\nX62888984Y36322278D01*\nX62817317Y36446408D01*\nX62728162Y36505273D01*\nX62662896Y36522761D01*\nX62557772Y36515953D01*\nX62309511Y36372619D01*\nX62251054Y36284983D01*\nX62233566Y36219718D01*\nX62239967Y36113075D01*\nX62311633Y35988945D01*\nX61849967Y36788574D02*\nX61938566Y36730673D01*\nX62003832Y36713185D01*\nX62109511Y36719030D01*\nX62357772Y36862363D01*\nX62415674Y36950962D01*\nX62433162Y37016228D01*\nX62427317Y37121908D01*\nX62355651Y37246038D01*\nX62266496Y37304903D01*\nX62201230Y37322391D01*\nX62096106Y37315583D01*\nX61847845Y37172249D01*\nX61789387Y37084613D01*\nX61771899Y37019347D01*\nX61778300Y36912704D01*\nX61849967Y36788574D01*\nX61769853Y37974001D02*\nX61657772Y38074799D01*\nX61593062Y38091324D01*\nX61486419Y38084923D01*\nX61362289Y38013256D01*\nX61304388Y37924657D01*\nX61286344Y37860354D01*\nX61292745Y37753711D01*\nX61483300Y37423659D01*\nX62349325Y37923659D01*\nX62182659Y38212334D01*\nX62093504Y38271199D01*\nX62028238Y38288687D01*\nX61923114Y38281879D01*\nX61840360Y38234101D01*\nX61780940Y38145910D01*\nX61763452Y38080644D01*\nX61769853Y37974001D01*\nX61936520Y37685326D01*\nX60759411Y38677472D02*\nX61212631Y38939138D01*\nX61319274Y38945539D01*\nX61408429Y38886675D01*\nX61503429Y38722130D01*\nX61509829Y38615488D01*\nX60800788Y38701361D02*\nX60807189Y38594718D01*\nX60926078Y38388796D01*\nX61015233Y38329932D01*\nX61121875Y38336333D01*\nX61203666Y38383555D01*\nX61262530Y38472710D01*\nX61256130Y38579352D01*\nX61137241Y38785274D01*\nX61130840Y38891916D01*\nX60559411Y39023882D02*\nX61136762Y39 [... cut ...]
+X61695535Y40899185D01*
+X61614517Y40676592D01*
+X61611449Y40669793D01*
+X61498158Y40461756D01*
+X61494108Y40455485D01*
+X61351078Y40266660D01*
+X61346143Y40261067D01*
+X61176560Y40095661D01*
+X61170843Y40090865D01*
+X60978516Y39952591D01*
+X60972135Y39948691D01*
+X60761345Y39840626D01*
+X60754474Y39837729D01*
+X60529918Y39762282D01*
+X60522692Y39760442D01*
+X60289399Y39719306D01*
+X60281982Y39718564D01*
+X60045170Y39712658D01*
+X60045170Y31989949D01*
 X60374571Y32180129D01*
 X60378037Y32181952D01*

ktrace, 2022-10-20:
var/tmp/portage/sci-electronics/librepcb-0.1.7/work/librepcb-0.1.7/tests/unittests/core/project/board/boardplanefragmentsbuildertest.cpp:104: Failure
Expected equality of these values:
  expected.toStdString()
    Which is: "(actual\n (plane 007538f6-59ce-46d9-8a1f-f4b5cf3c73a7\n  (fragment\n   (vertex (position 19.198751 4.01995) (angle 0.0))\n   (vertex (position 19.389572 4.079004) (angle 0.0))\n   (vertex (position 19.564849 4.174806) (angle 0.0))\n   (vertex (position 19.717589 4.303533) (angle 0.0))\n   (vertex (position 19.841696 4.460049) (angle 0.0))\n   (vertex (position 19.93222 4.638109) (angle 0.0))\n   (vertex (position 19.985549 4.830608) (angle 0.0))\n   (vertex (position 20.0 5.0) (angle 0.0))\n   (vertex (position 20.0 9.1872) (angle 0.0))\n   (vertex (position 19.98005 9.385951) (angle 0.0))\n   (vertex (position 19.920996 9.576772) (angle 0.0))\n   (vertex (position 19.825194 9.752049) (angle 0.0))\n   (vertex (position 19.696467 9.904789) (angle 0.0))\n   (vertex (position 19.539951 10.028896) (angle 0.0))\n   (vertex (position 19.361891 10.11942) (angle 0.0))\n   (vertex (position 19.169392 10.172749) (angle 0.0))\n   (vertex (position 19.0 10.1872) (angle 0.0))\n   (vertex (position 18.029177 10.1872) (angle 0.0))\n 
...
  actual.toStdString()
    Which is: "(actual\n (plane 007538f6-59ce-46d9-8a1f-f4b5cf3c73a7\n  (fragment\n   (vertex (position 19.198751 4.01995) (angle 0.0))\n   (vertex (position 19.389572 4.079004) (angle 0.0))\n   (vertex (position 19.564849 4.174806) (angle 0.0))\n   (vertex (position 19.717589 4.303533) (angle 0.0))\n   (vertex (position 19.841696 4.460049) (angle 0.0))\n   (vertex (position 19.93222 4.638109) (angle 0.0))\n   (vertex (position 19.985549 4.830608) (angle 0.0))\n   (vertex (position 20.0 5.0) (angle 0.0))

With diff:
@@ -549,32 +549,32 @@
    (vertex (position 77.406254 60.719104) (angle 0.0))
    (vertex (position 77.233219 60.75) (angle 0.0))
-   (vertex (position 65.076989 60.75) (angle 0.0))
-   (vertex (position 64.937332 60.7301) (angle 0.0))
-   (vertex (position 64.808792 60.671984) (angle 0.0))
-   (vertex (position 64.701601 60.580278) (angle 0.0))
-   (vertex (position 64.62429 60.462282) (angle 0.0))
-   (vertex (position 64.583014 60.327387) (angle 0.0))
-   (vertex (position 64.554832 60.1475) (angle 0.0))
-   (vertex (position 64.546915 60.113093) (angle 0.0))
-   (vertex (position 64.515994 60.01204) (angle 0.0))
-   (vertex (position 64.503303 59.979098) (angle 0.0))
-   (vertex (position 64.398549 59.755708) (angle 0.0))
-   (vertex (position 64.381327 59.724864) (angle 0.0))
-   (vertex (position 64.323426 59.636493) (angle 0.0))
-   (vertex (position 64.302021 59.608382) (angle 0.0))
-   (vertex (position 64.139053 59.423156) (angle 0.0))
-   (vertex (position 64.113904 59.398353) (angle 0.0))
-   (vertex (position 64.033623 59.329669) (angle 0.0))
-   (vertex (position 64.005224 59.30866) (angle 0.0))
-   (vertex (position 63.796989 59.176314) (angle 0.0))
-   (vertex (position 63.765925 59.159529) (angle 0.0))
-   (vertex (position 63.669636 59.115994) (angle 0.0))
-   (vertex (position 63.63652 59.103761) (angle 0.0))
-   (vertex (position 63.399595 59.034842) (angle 0.0))
-   (vertex (position 63.36509 59.027404) (angle 0.0))
-   (vertex (position 63.251706 59.011248) (angle 0.0))
-   (vertex (position 63.234119 59.009374) (angle 0.0))
-   (vertex (position 63.110722 59.000627) (angle 0.0))
-   (vertex (position 63.093026 59.0) (angle 0.0))
+   (vertex (position 65.076987 60.75) (angle 0.0))
+   (vertex (position 64.93733 60.7301) (angle 0.0))
+   (vertex (position 64.80879 60.671984) (angle 0.0))
+   (vertex (position 64.701599 60.580278) (angle 0.0))
+   (vertex (position 64.624288 60.462282) (angle 0.0))
+   (vertex (position 64.583012 60.327387) (angle 0.0))
+   (vertex (position 64.55483 60.1475) (angle 0.0))
+   (vertex (position 64.546913 60.113093) (angle 0.0))
+   (vertex (position 64.515992 60.01204) (angle 0.0))
+   (vertex (position 64.503301 59.979098) (angle 0.0))
+   (vertex (position 64.398547 59.755708) (angle 0.0))
+   (vertex (position 64.381325 59.724864) (angle 0.0))
+   (vertex (position 64.323424 59.636493) (angle 0.0))
+   (vertex (position 64.302019 59.608382) (angle 0.0))
+   (vertex (position 64.139051 59.423156) (angle 0.0))
+   (vertex (position 64.113902 59.398353) (angle 0.0))
+   (vertex (position 64.033621 59.329669) (angle 0.0))
+   (vertex (position 64.005222 59.30866) (angle 0.0))
+   (vertex (position 63.796987 59.176314) (angle 0.0))
+   (vertex (position 63.765923 59.159529) (angle 0.0))
+   (vertex (position 63.669634 59.115994) (angle 0.0))
+   (vertex (position 63.636518 59.103761) (angle 0.0))
+   (vertex (position 63.399593 59.034842) (angle 0.0))
+   (vertex (position 63.365088 59.027404) (angle 0.0))
+   (vertex (position 63.251704 59.011248) (angle 0.0))
+   (vertex (position 63.234117 59.009374) (angle 0.0))
+   (vertex (position 63.11072 59.000627) (angle 0.0))
+   (vertex (position 63.093024 59.0) (angle 0.0))
    (vertex (position 26.504863 59.0) (angle 0.0))
    (vertex (position 26.495148 59.000189) (angle 0.0))
[... 32 lines cut ...]
+   (vertex (position 63.311385 60.959688) (angle 0.0))
+   (vertex (position 63.301942 60.96514) (angle 0.0))
+   (vertex (position 63.266309 60.99009) (angle 0.0))
+   (vertex (position 63.257951 60.997104) (angle 0.0))
+   (vertex (position 63.223341 61.02366) (angle 0.0))
[... cut ...]
+   (vertex (position 63.108385 61.123848) (angle 0.0))
    (vertex (position 58.788695 65.443538) (angle 0.0))
    (vertex (position 58.675872 65.528219) (angle 0.0))
@@ -2204,6 +2204,6 @@
    (vertex (position 26.422612 60.006025) (angle 0.0))
    (vertex (position 26.5 60.0) (angle 0.0))
-   (vertex (position 63.101882 60.0) (angle 0.0))
-   (vertex (position 63.241539 60.0199) (angle 0.0))
+   (vertex (position 63.10188 60.0) (angle 0.0))
+   (vertex (position 63.241537 60.0199) (angle 0.0))
   )
   (fragment

ktrace, 2022-10-20:
Please note:
```
-   (vertex (position 63.101882 60.0) (angle 0.0))
-   (vertex (position 63.241539 60.0199) (angle 0.0))
+   (vertex (position 63.10188 60.0) (angle 0.0))
+   (vertex (position 63.241537 60.0199) (angle 0.0))
```
Look like issue #974

ubruhin, 2022-10-20:
Hi,

Did you see a concrete problem or is the unit test failure the only problem? The Gerber export test is very picky and fails even if a plane coordinate is 1 nanometer different than the expected value. This is a known "problem", see #516. But it's not a problem in practice since the deviations are very small. This is just due to different compiler optimizations etc. which can lead to slightly different mathematical calculation results.

If there is no concrete bug, I'd close this issue since we have #516 to keep this in mind.