LibrePCB/LibrePCB #1058: Register *.lpp files from within the application

ubruhin, 2022-10-25:
##### SUMMARY

Currently the `*.lpp` file extension is registered with the LibrePCB application only when using the installer on Windows or Linux. But with the portable packages (e.g. the AppImage) this is not the case. That's the reason why the installer is promoted as the recommended installation method, although this is quite unusual on Linux.

##### SOLUTION

It would be nice if the application itself can (optionally) register the file extension `*.lpp`. Then the portable packages would be much more useful.

However, there should be many things to consider carefully:
* The application might run with user rights, possibly not able to register a file extension (?)
* If the installation method already registered the file extension (maybe still done in the installer? maybe done by a Linux package management like Snap/Flatpak?) the option to register the file extension from within the application should be disabled.
* Registering the file extension shall be optional. But where to put the option?
  * In the wizard which appears when starting LibrePCB the first time?
  * In the workspace settings dialog?
  * As a (dismissable) popup in the control panel?
 * Somewhere also provide an option to unregister the extension.
 * Can we somehow also provide this feature on macOS?

Once this is implemented (and working with the AppImage), probably we should not promote the installer for Linux anymore.

t2ripit, 2023-02-19:
I primarily use macOS so I looked into this. The Info.plist file inside the app bundle contains the information for registering file extensions. Unfortunately there is a conflict, at least with Xcode installed, it has already registered `.lpp` with the lex-source UTI. 

Assuming the project is committed to the lpp extension, there may not be a automatic solution for macos when Xcode is also installed. For users without Xcode installed, this should be doable, but I will have to look into how the build system generates the Info.plist file.

ubruhin, 2023-02-19:
Great, thanks for looking into this @t2ripit! I was not aware of this conflict with our file extension, but so far I do not plan to change it (there's always a risk of conflicting with some other application...).

ubruhin, 2023-04-08:
@t2ripit FYI, in the mean time I extended the `Info.plist` file by myself: https://github.com/LibrePCB/LibrePCB/pull/1145 

Although I couldn't get it working with the recommended `LSItemContentTypes` key, at least it seems to work so far. However I don't know what happens if Xcode is also installed. If you like to try it, here's a *.dmg to test (unstable file format, don't actually use the application!): https://download.librepcb.org/nightly_builds/macos-register-file-extension/librepcb-nightly-mac-x86_64.dmg

PR #1146 (2023-04-10): Linux: Support registering *.lpp file extension at runtime
This feature allows to (optionally) integrate portable Linux packages (mainly the AppImage) into the desktop environment by installing the mime types & *.desktop file. This has always been a big drawback of our AppImage, which is finally fixed.

The option is available in the control panel (if applicable) and in the workspace settings dialog:

[image]

To avoid feeling it like "black magic", a dialog lists all files which will be created or deleted by this feature.

Note that currently this feature is only available in our official Lin [... cut]
closed 2023-04-10 by ubruhin