LibrePCB/LibrePCB #1069: Gerber generation missing from menu

abulka, 2022-11-28:
Gerber generation missing from menu

> The manual says this is possible: Gerber/Excellon Open File › Production Data › Generate Fabrication Data or press F10 . Then choose the settings of the Gerber/Excellon export.

```
LibrePCB Version: 0.1.7
Git Revision:     ba553c6fb
Build Date:       2022-10-03T08:32:18
Qt Version:       5.15.6 (built against 5.15.6)
CPU Architecture: x86_64
Operating System: macOS 13.0
Platform Plugin:  cocoa
TLS Library:      Secure Transport, macOS 13.0
```
Did I miss an installation step or something?

[image]

ubruhin, 2022-11-29:
The menu item exists only in the board editor, not in the schematic editor. So probable the focus was on the wrong window.

abulka, 2022-11-29:
Oh, I see - thanks!

closed 2022-11-29 by abulka