LibrePCB/LibrePCB #1073: Board editor: Trace layer sometimes not respected on THT pads

ubruhin, 2022-12-19:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 0.1.7
Git Revision:     ba553c6
Build Date:       2022-10-03T08:30:22
Qt Version:       5.15.2 (built against 5.15.2)
CPU Architecture: x86_64
Operating System: Manjaro Linux
Platform Plugin:  xcb
TLS Library:      OpenSSL 1.1.1l  24 Aug 2021
```

##### SUMMARY

When starting a new trace on a THT pad, the selected layer (in the toolbar) is not respected if there's already a trace on any other layer on the same pad - then the same layer will be used for the new trace. Thus it seems to be impossible to add traces on multiple layers to the same THT pad. Fortunately there's a workaround: Just hide the layer of the existing trace, then the selected layer will be used for the new trace.

##### STEPS TO REPRODUCE

1. Add a device with THT pads to the board.
2. Connect a trace on the top layer to the THT pad.
3. In the "draw trace" toolbar, select the bottom layer.
4. Start drawing a new trace on the same pad.

##### EXPECTED RESULTS

The new trace is drawn on the bottom layer.

##### ACTUAL RESULTS

The new trace is drawn on the top layer.

t2ripit, 2023-02-17:
I looked into this, the cause is that `BoardEditorState::findItemsAtPos` gives "top netlines" higher priority over "top pads", so the new trace segment is starting on the netline on the top layer instead of the pad. When the new trace starts on a netline, the layer selection automatically changes to the layer of the existing netline. 

By adding the optional layer argument to the `findItemsAtPos` function call within `BoardEditorState_DrawTrace::startPositioning` this behavior is fixed. However, you can no longer start a trace on an existing netline on another layer, it simply starts a new trace on the layer selected in the toolbar. I think this is still preferable, but maybe not.

In my opinion the priority system in `findItemsAtPos` could be extended somewhat to improve the experience. But I'm sure a discussion should be had before making that kind of change. I also think the pad representation could be extended, possibly similar to another issue I saw on via padstacks.

ubruhin, 2023-02-18:
Thanks for looking into this @t2ripit. I wonder, wouldn't it make more sense to simply give THT pads a higher priority than netlines? I just saw that vias (i.e. almost the same thing as a THT pad) does have the highest possible priority. Maybe I simply forgot to distinguish between SMT pads and THT pads - the current priority logic seems to be reasonable only for SMT pads.

At least this would be a very simple fix and I guess it would behave as most users probably expect. However, "simple" in this editor logic does actually not exist, sometimes very subtle, innocent looking changes can lead to very nasty UX issues :see_no_evil:

t2ripit, 2023-02-19:
Giving through-hole pads higher priority should work. I can give it a shot and make sure there aren't unintended consequences.

ubruhin, 2023-02-19:
That would be great! :+1:

PR #1302 (2024-02-14): Board editor: Fix layer change when starting trace on THT pad
Fixes #1073