LibrePCB/LibrePCB #1078: Add support for plated slots in boards

ubruhin, 2023-01-05:
##### SUMMARY

Non-plated slots were implemented in #1071. Plated footprint pad slots were implemented in #1076. But pads can only be added to footprints (i.e. in the library editor), not directly to a board. However, I'm pretty sure sooner or later the need for plated slots in boards will arise as well, which is currently still not possible (except by wrapping it in a package, which is cumbersome).

Well, actually it would be possible to create plated slots by drawing polygons on the "Milling" layer. But since the polygon outline can be arbitrary, it's handled differently in the Gerber export (not part of drill files, separate milling file needs to be explained manually to the manufacturer, and a larger tool diameter might be required). It would be much better to create true slots which get exported to drill files.

##### SOLUTION

I'm really not sure how this should be implemented. Maybe extending via objects to allow creating slots? However, once blind/buried vias will be supported as well, I don't know if it's a good idea to support slots at the same time (i.e. blind/buried plated slots :sob:). In addition, vias currently can only be square-sized (round/square/octagon) which wouldn't work for slots.

Adding a "plated" property to hole objects is also strange, since holes are non-electrical objects but plated slots usually are connected to some net.

And allowing to add footprint pad objects to boards directly is also somewhat strange (would be mostly redundant to vias and thus confusing, while adding much more complexity).

Any ideas welcome ;-)

eto-, 2023-10-29:
Any news?
I need plated holes: I tried to use vias with the right size but they are not exported in the step file.

ubruhin, 2023-10-29:
No news here. As a workaround, you could create a dummy package consisting just of a pad with the desired size and add it to the schematic/board.

PR #1596 (2025-09-10): Support adding individual pads directly to boards
The initial goal of this was to support adding plated slots to boards as requested in #1078. The solution I have chosen is to allow adding arbitrary, individual footprint pads directly to boards. Footprint pads already support configuring them as slots, so this feature implicitly allows to create slots in boards.

Unfortunately this solution is a bit complex (affecting file format, UI, FSM, DRC, Gerber export, PnP export, Netlist export, Specctra export/import). But I think it is worth the complexity because it allows not only to add plated slo [... cut]