LibrePCB/LibrePCB #1091: Please support older macOS versions

julian-weinert, 2023-01-31:
Today I discovered that the current LibrePCB version only supports macOS Big Sur.
Especially in this industry I find people running older OS versions, including me.

I compiled it myself and it is indeed compatible down to Mojave.
As far as I know it would suffice to add `set(CMAKE_OSX_DEPLOYMENT_TARGET "10.14" CACHE STRING "Minimum OS X deployment version")` to the CMakeLists.txt.

Thanks for the consideration and great work :)

ubruhin, 2023-01-31:
Thanks for the suggestion. I was not aware of `CMAKE_OSX_DEPLOYMENT_TARGET` - if that works, that would be really nice.

I quickly implemented this on a branch (662b9a2c289c6ebba4f40b3eee2be494336c6127) but since I don't have macOS I cannot test it. @julian-weinert could you maybe verify that:

- This nightly build with the fix runs on macOS 10.14: https://download.librepcb.org/nightly_builds/1091-older-osx-deployment-target/librepcb-nightly-mac-x86_64.dmg
- And optionally for reference, this nightly build without the fix does *not* run on macOS 10.14: https://download.librepcb.org/nightly_builds/master/librepcb-nightly-mac-x86_64.dmg

That would be very helpful. Just don't save anything to disk with these nightly builds, they use an unstable file format. I recommend to temporarily use a different workspace (e.g. just rename your workspace folder before starting LibrePCB).

julian-weinert, 2023-02-01:
@ubruhin I'm happy to report back it works! The first one runs, the other doesn't :)

spookydd, 2023-02-01:
> 
> 
> Would you maybe compile one version lower, 10.13 as well? Then we can test how far it goes down

I think that would be a good thing to do.

I’m still on 10.13 myself and have been out of this game since the support was basically dropped, and I’m not about to upgrade my macOS any time soon.

So it would be nice if I could again do a simple install and try out some things.

Thanks guys,

ubruhin, 2023-02-01:
> I'm happy to report back it works!

Awesome! :tada: 

> Would you maybe compile one version lower, 10.13 as well?

Done - the first download link above now points to the binary compiled for 10.13.

But lower than 10.13 is probably not needed, is it?

julian-weinert, 2023-02-01:
Here it didn't even launch, just crapped out immediately. I would agree. 10.13 is already pretty low balled :P
Couldn't you actually test that yourself then, when you're on 10.13 still? :D

Thanks for the effort

julian-weinert, 2023-02-01:
I think you may not have properly cleaned, is that possible? We got this in the crash log
```
Dyld Error Message:
  Symbol not found: ____chkstk_darwin
  Referenced from: librepcb.app/Contents/MacOS/../Frameworks/libpcre2-16.0.dylib (which was built for Mac OS X 11.0)
```

ubruhin, 2023-02-01:
Hmm OK so maybe we should stick with 10.14 for now (sorry @spookydd). Looks like a bundled shared library accesses a function not provided in macOS 10.13.

> I think you may not have properly cleaned, is that possible?

No that's not possible since it was a clean build [on CI](https://dev.azure.com/LibrePCB/LibrePCB/_build). The CI is also the reason why I had to switch to macOS 11 - Microsoft dropped the macOS 10 build agent. I always try to build on older systems to get the highest compatibility of the binaries, but for macOS and Windows I am totally dependent on what Azure provides :sob:

PR #1099 (2023-02-08): Support older macOS versions (>=10.14)
Set `CMAKE_OSX_DEPLOYMENT_TARGET` to macOS 10.14 to make our official binaries running on older macOS versions again.

Closes #1091
closed 2023-02-08 by ubruhin