LibrePCB/LibrePCB #1092: DRC: need to disable "Pad is not connected to any signal."

hooperbloob, 2023-01-31:
Trying to create a simple breakout board with a connector so it has no real signals, just connections to larger solder pads. Is there a way we can optionally disable the DRC that prevents creating these pads?  I'm blocked by "Pad is not connected to any signal."

ubruhin, 2023-01-31:
Hm, so you want to connect two pads with a trace? Why don't you create this connection in the schematic first?

This error message does not come from the DRC - it's a fundamental concept of LibrePCB that boards are always in sync with the schematics. This error prevents you from breaking this concept as your trace does not correspond to a connection in the schematics.

So I'd suggest to draw a schematic as usual, then you will be able to draw the trace. Or maybe post some more details/screenshots what you want to do exactly :slightly_smiling_face:

hooperbloob, 2023-01-31:
The problem is that the solder pads aren't components per se, not going to appear in the BOM, they're just artifacts on the PCB.

I followed your suggestion and added a pair of 0.1" pin headers and connected those but those just exist to satisfy the tool.

Do we need to add some non-component components to the library?

[image]
[image]

ubruhin, 2023-01-31:
> The problem is that the solder pads aren't components per se, not going to appear in the BOM, they're just artifacts on the PCB.

I'm not sure, IMHO usually one adds and connects solder pads in the schematic as well since they have an electrical purpose (like test points or connectors). I agree these should not appear in the BOM, this is a configuration we'll add to LibrePCB some day.

> Do we need to add some non-component components to the library?

Probably yes, but I'd expect this to be used only for pure mechanical (i.e. non-electrical) components. Adding electrical connections without LibrePCB knowing about their electrical purpose sounds like a nightmare (might lead to many nasty bugs or unexpected behavios). Just think about the DRC clearance check which doesn't know the nets of the traces...

For your use-case of a breakout board, I'd say the best option would be to implement the [reverse engineering mode](https://librepcb.discourse.group/t/reverse-engineering-mode/21) (like a back-annotation). This would allow you to define the electrical connections by connecting traces in the board instead of in the schematic. Drawing a schematic would be optional in that case, thus not wasting any time for simple/hacky PCBs. As this is already an open feature request, I'll close this issue. Please comment/reopen if you disagree :slightly_smiling_face: