LibrePCB/LibrePCB #1094: Board editor: usability - hard to select devices placed near the 'corners' of larger holes

hooperbloob, 2023-02-04:
##### SUMMARY
The mouse click detection area for a hole placed in a pcb seems to be a bounding square. This is fine for small holes 
but if you place devices in the corner regions of larger ones they cannot be selected via the mouse, you end up 
selecting the hole instead.

##### SOLUTION
Refine the click detection area for holes to keep within a few pixels of the hole border.

ubruhin, 2023-02-04:
Thanks for reporting, I can reproduce this with LibrePCB 0.1.7 indeed. However, with the current `master` it's working as expected, probably this was fixed during the implementation of slotted holes.

Therefore closing this issue. We only need to wait for the next release :see_no_evil: This is the kind of issues I like :grin:

closed 2023-02-04 by ubruhin

dbrgn, 2023-02-05:
> probably this was fixed during the implementation of slotted holes

For the record, that was #1071