LibrePCB/LibrePCB #1096: Board editor: provide simple csv import/export for border vertices or better copy/paste

hooperbloob, 2023-02-04:
##### SUMMARY
Entering the x-y points for a board with large curved sections is a pain. It'd be nice if I could copy/paste them from an external spreadsheet or import via a CSV file.  The current table widget only allows you to select single rows. If I could select all or most of them then I could copy them to the clipboard for group editing elsewhere.

##### SOLUTION
Allow the user to select multiple rows in the border vertex widget. Implement copy/paste for this.

ubruhin, 2023-02-07:
Just out of curiosity, where would you have the spreadsheet from? Manually written, or exported from a mechanical CAD?

Currently you can already import a DXF which might be even easier as long as you have the board outline already available in a MCAD.

hooperbloob, 2023-02-07:
The board I'm creating has its right border in the shape of a semi-circle (like a big 'D') so I wrote a script to generate the points needed and then entered those manually into LibrePCB. 

If LibrePCB supported copy & paste I'd have formatted it as CSV, copied it to my clipboard, and then pasted it into that table widget. I mentioned a spreadsheet as a starting point if someone were to do the same while re-scaling or adjusting the offset repeatedly.

Tubbles, 2023-02-13:
> so I wrote a script to generate the points needed

What format is it in originally?