LibrePCB/LibrePCB #1170: Board editor: be able to group items together to make it easier to move them jointly without deforming

kotchin, 2023-06-12:
##### SUMMARY
When working in the board editor and there are multiple items (such as lines) which need to be moved, it can be challenging to both grab them as a group but also move them without deforming them. 

##### SOLUTION
There may be more than one way to solve it, but what I was intuitively looking for was a way to group them together into one entity and drag/drop them all together, but I kept grabbing lines which when dragged/dropped were modified/deformed. This is a convenience feature but could be helpful nonetheless.

If I have missed where this is in the settings, documented or in the issues, I apologize. I did not find it myself.

Below an example of what I was working with when I was looking for this feature.

[image]

ubruhin, 2023-06-13:
With "deforming", you mean moving a single polygon vertex instead of the whole object, right?

I think the simplest and most intuitive way to fix this is to enable vertex movement only if a single object is selected. If multiple objects are selected, vertex movement makes no sense so only the whole objects should be moved then.

kotchin, 2023-06-13:
> With "deforming", you mean moving a single polygon vertex instead of the whole polygon, right?

Precisely.

> I think the simplest and most intuitive way to fix this is to enable vertex movement only if a single object is selected. If multiple objects are selected, vertex movement makes no sense so only the whole objects should be moved then.

That sounds like an intuitive solution which would also solve it, indeed. It would remove the need for additional settings and would be more in line with how I guess people would expect the polygons to behave.

closed 2026-10-30 by ubruhin