LibrePCB/LibrePCB #1179: Select-all in library editor only selects items within 14 squares of the center

kotchin, 2023-07-19:
Thank you for creating LibrePCB. I really enjoy using it.

I'm sorry to have to report a bug. This probably qualifies as a minor bug.

##### SUMMARY
When using CTRL+A to select all in the library editor, the pins, shapes and lines being selected are only those within 14 squares of the center (vertically and horizontally). There is no issue selecting any/all of them when using the mouse.
This bug does not occur in the schematic editor.

##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 0.1.7
Git Revision:     ba553c6
Build Date:       2022-10-03T14:10:33
Qt Version:       5.15.10 (built against 5.15.6)
CPU Architecture: x86_64
Operating System: KDE Flatpak runtime
Platform Plugin:  wayland
TLS Library:      OpenSSL 3.0.8 7 Feb 2023
```

##### EXPECTED RESULTS
All items and pins should be selected.

##### ACTUAL RESULTS
Only pins within 14 squares of the center are selected, horizontally and vertically.

##### ADDITIONAL INFORMATION
Below is the result of using CTRL+A, with the symbol in 3 various positions (notice how the selected pins differ in each case).

[image]

[image]

[image]

[image]

I'm not that great at programming, but if you need more help with identifying the bug, I can see what I can do.

ubruhin, 2023-07-19:
> Thank you for creating LibrePCB. I really enjoy using it.

Glad to hear! :heart: 

> When using CTRL+A to select all in the library editor, the pins, shapes and lines being selected are only those within 14 squares of the center (vertically and horizontally). 

Oh that's indeed a bug, I was able to reproduce it!  Thanks for reporting with all the details! :+1:

BuFran, 2023-09-29:
the bug is inside libs/librepcb/editor/library/sym/fsm/symboleditorstate_select.cpp on line 326. 

I'm investigating possible solution now ...

BuFran, 2023-10-02:
Please @ubruhin check if it is still reproducible. 

Looking in code, trying many symbols and I still didn't reproduced the bug. From the code, it seems everything is fine and maybe it has been corrected in some of previous PR's.

If it's still reproducible on master, please tell me which symbol I should use for test. Same code is used for Schematic and Board editor, so the same bug should be visible there.

The link above from my comment has been "entrypoint" where to start searching consequences, not real description of bug.

ubruhin, 2023-10-02:
I can still reproduce it. For example in the ST Microelectronics library, open the symbol "STM32C0 IO-45 MonoIO-1 Power-2" and press Ctrl+A. This selects only pins close to the origin, but not those far away.

BuFran, 2023-10-02:
Acknowledged that `mContext.symbolGraphicsItem` is empty (no items). 

I'm looking now (inside schematic editor sources) where they should be filled. Seems there is missing the fill/copy code somewhere I missed that.

BuFran, 2023-10-02:
Fixed for me (I'm preparing PR now), but for me, there isn't very clear meaning of `symbolGraphicsItem` for that case. 

I need to dig into architecture to fully understand the meaning of the class. Seems that is result of partial unfinished refactoring and works write-only now.

BuFran, 2023-10-02:
Same bug is in package editor