LibrePCB/LibrePCB #1205: aarch64 build fails

Alexander-Wilms, 2023-08-22:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version:
Git Revision: 1.0.0-rc1
Qt Version: 5.15
Operating System: Flatpak runtime org.kde.Platform//5.15-22.08
```

##### SUMMARY
The aarch64 build of version 1.0.0-rc1 fails while the x86_64 build succeeds.

##### STEPS TO REPRODUCE
Run `flatpak-builder --force-clean --ccache --repo=librepcb-repo librepcb-build org.librepcb.LibrePCB` with the following manifest:
```
{
    "app-id": "org.librepcb.LibrePCB",
    "runtime": "org.kde.Platform",
    "runtime-version": "5.15-22.08",
    "sdk": "org.kde.Sdk",
    "command": "librepcb",
    "finish-args": [
        "--device=dri",
        "--socket=x11",
        "--socket=wayland",
        "--share=ipc",
        "--share=network", /* required for downloading libraries */
        "--filesystem=home" /* required for storing workspace, libraries etc. */
    ],
    "modules": [
        "shared-modules/glu/glu-9.json",
        {
            "name": "librepcb",
            "buildsystem": "cmake-ninja",
            "builddir": true,
            "config-opts": [
                "-DUSE_OPENCASCADE=0"
            ],
            "sources": [
                {
                    "type": "git",
                    "url": "https://github.com/LibrePCB/LibrePCB",
                    "tag": "1.0.0-rc1",
                    "commit": "2d9ea3c230ef8aa5c431216a3daac829b56a617c"
                }
            ]
        }
    ]
}
```

##### EXPECTED RESULTS
I expext the build to succeed.

`[681/833] Building CXX object libs/librepcb/editor/CMakeFiles/librepcb_editor.dir/widgets/openglview.cpp.o`

##### ACTUAL RESULTS
```
[660/833] Building CXX object libs/librepcb/editor/CMakeFiles/librepcb_editor.dir/widgets/openglview.cpp.o
FAILED: libs/librepcb/editor/CMakeFiles/librepcb_editor.dir/widgets/openglview.cpp.o 
/usr/bin/c++ -DQT_CONCURRENT_LIB -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_PRINTSUPPORT_LIB -DQT_QMLMODELS_LIB -DQT_QML_LIB -DQT_QUICKCONTROLS2_LIB -DQT_QUICK_LIB -DQT_SQL_LIB -DQT_SVG_LIB -DQT_WIDGETS_LIB -DTYPE_SAFE_ARITHMETIC_POLICY=1 -DTYPE_SAFE_ENABLE_ASSERTIONS=0 -DTYPE_SAFE_ENABLE_PRECONDITION_CHECKS=1 -DTYPE_SAFE_ENABLE_WRAPPER=1 -I/run/build/librepcb/_flatpak_build/libs/librepcb/editor -I/run/build/librepcb/libs/librepcb/editor -I/run/build/librepcb/_flatpak_build/libs/librepcb/editor/librepcb_editor_autogen/include -I/run/build/librepcb/libs/librepcb/editor/../.. -I/run/build/librepcb/_flatpak_build/libs/librepcb/core -I/run/build/librepcb/libs/optional/tl -I/run/build/librepcb/libs/type_safe/include -I/run/build/librepcb/libs/type_safe/external/debug_assert -isystem /usr/include/QtConcurrent -isystem /usr/include/QtCore -isystem /usr/lib/mkspecs/linux-g++ -isystem /usr/include/QtGui -isystem /usr/include/QtNetwork -isystem /usr/include/QtPrintSupport -isystem /usr/include/QtWidgets -isystem /usr/include/QtSql -isystem /usr/include/QtSvg -isystem /usr/include/QtOpenGL -isystem /usr/include/QtQuick -isystem /usr/include/QtQmlModels -isystem /usr/include/QtQml -isystem /usr/include/QtQuickControls2 -O2 -g -pipe -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -fasynchronous-unwind-tables -fstack-clash-protection -fno-omit-frame-pointer -O3 -DNDEBUG -std=c++11 -Wall -Wextra -Wpedantic -Wno-noexcept-type -Wno-deprecated-declarations -fPIC -MD -MT libs/librepcb/editor/CMakeFiles/librepcb_editor.dir/widgets/openglview.cpp.o -MF libs/librepcb/editor/CMakeFiles/librepcb_editor.dir/widgets/openglview.cpp.o.d -o libs/librepcb/editor/CMakeFiles/librepcb_editor.dir/widgets/openglview.cpp.o -c /run/build/librepcb/libs/librepcb/editor/widgets/openglview.cpp
/run/build/librepcb/libs/librepcb/editor/widgets/openglview.cpp: In member function ‘virtual void librepcb::editor::OpenGlView::initializeGL()’:
/run/build/librepcb/libs/librepcb/editor/widgets/openglview.cpp:242:12: error: ‘GL_MULTISAMPLE’ was not declared in this scope; did you mean ‘GL_MULTISAMPLES_NV’?
  242 |   glEnable(GL_MULTISAMPLE);
      |            ^~~~~~~~~~~~~~
      |            GL_MULTISAMPLES_NV
/run/build/librepcb/libs/librepcb/editor/widgets/openglview.cpp:245:12: error: ‘GL_LINE_SMOOTH’ was not declared in this scope; did you mean ‘GL_LINE_LOOP’?
  245 |   glEnable(GL_LINE_SMOOTH);
      |            ^~~~~~~~~~~~~~
      |            GL_LINE_LOOP
/run/build/librepcb/libs/librepcb/editor/widgets/openglview.cpp:246:10: error: ‘GL_LINE_SMOOTH_HINT’ was not declared in this scope
  246 |   glHint(GL_LINE_SMOOTH_HINT, GL_NICEST);
      |          ^~~~~~~~~~~~~~~~~~~
```

##### ADDITIONAL INFORMATION

ubruhin, 2023-08-22:
Perfect, thanks for the report! :+1: Added to the 1.0.0 milestone since this is something we really need to fix for the stable release.

Alexander-Wilms, 2023-08-22:
Possible fix:

https://github.com/colmap/colmap/issues/745#issuecomment-766476624

PR #1208 (2023-08-26): OpenGlView: Fix compile error on missing OpenGL defines
Manually adding some OpenGL defines not available on some targets. I doubt this is the right way to fix it, but I didn't find a better solution yet...

Fixes #1205
closed 2023-08-26 by ubruhin