LibrePCB/LibrePCB #1217: All editors : BUG : problem with right click

doom-fr, 2023-08-28:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 1.0.0-rc1
Git Revision:     2d9ea3c23
Build Date:       2023-08-21T13:03:25
Build Author:     LibrePCB CI
Qt Version:       5.15.2 (built against 5.15.2)
CPU Architecture: x86_64
Operating System: Ubuntu 23.04
Platform Plugin:  xcb
TLS Library:      OpenSSL 1.1.1t  7 Feb 2023
Runtime:          AppImage
```

##### SUMMARY
Right click does not works properly, context menu key also

##### STEPS TO REPRODUCE
right click on a symbol in schematic or on a trace in board editor !

##### EXPECTED RESULTS
open the contextual menu

##### ACTUAL RESULTS

[image]
Sorry for the video, but it cannot be recorded correctly in video screenshot.

##### ADDITIONAL INFORMATION
also problem with the contextual key on the keyboard
might be related to the use of QT as I experienced this also in Freecad but not for example in Nautilus.
might also be related to AppImage
https://forum.freecad.org/viewtopic.php?t=80010

ubruhin, 2023-08-28:
Huh, that's not good :( I never experienced that problem, not even with the AppImage on Ubuntu 20.04 and 22.04 (didn't test on 23.04 yet).

May I ask what device you are using, touchpad/buttons of a notebook, or external keyboard/mouse on a desktop?

And does the 0.1.7 AppImage work correctly on the same machine?

doom-fr, 2023-08-28:
I am using a notebook with external keyboard and mouse, but the behavior is the same : 
- with 0.1.7
- if I switch off the touchpad
- switch back on and use the touchpad
- disconnect external mouse, keyboard, screen and use the touchpad
- use the installer version

Right click work perfectly in Firefox, Nautilus, terminal, ...

Do you know another application on Ubuntu that uses QT ? To have a try ...

ubruhin, 2023-08-28:
Ok, thanks for the information! So it does not seem to be related to the 1.0 release.

In your link to the FreeCad forum there's a post that it *could* be related to Wayland, maybe you could check if the issue doesn't exist when switching to X11 (AFAIK at least on Ubuntu 22.04 this is at the bottom right of the login screen)?

> Do you know another application on Ubuntu that uses QT ? To have a try ...

I think KeePass(XC), WireShark, LibreCAD, MuseScore are written in Qt. See also https://en.wikipedia.org/wiki/Category:Software_that_uses_Qt.

doom-fr, 2023-08-29:
Good news, switching to X11 solve the problem in LibrePCB and also in Freecad. 
Unfortunately, it brings also other major problems with video files. :sob:

ubruhin, 2023-08-29:
Okay thanks for confirmation. This is very strange, here with Wayland on Ubuntu 22.04 the AppImage works fine.

To be honest, I'm not sure if we can fix that somehow. It seems to be a very specific issue (possibly not related to LibrePCB at all) and I have no idea how to debug that as long as I cannot reproduce it...

Since you're using Ubuntu, maybe the LibrePCB Snap (from the Ubuntu Store) works for you. At least it's worth a try.

doom-fr, 2023-08-29:
Snap did not help.

doom-fr, 2023-08-29:
In LibrePCB logs, I have : 
```
./librepcb-1.0.0-rc1-linux-x86_64.AppImage 
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[  INFO   ] LibrePCB 1.0.0-rc1 (2d9ea3c23)
[  INFO   ] Qt version: 5.15.2 (compiled against 5.15.2)
```

What do you think of the warning ?

Is there a way to have more verbose logs ?

ubruhin, 2023-08-29:
Ok I see. The thing is, for the Snap we explicitly disable Wayland to make LibrePCB running with Xwayland (X11 emulation on Wayland) mode because several things are known to be not working on Wayland. For the AppImage I guess we do not disable Wayland but it might be disabled implicitly by not containing the Wayland platform plugin. Maybe the warning is coming from there. And maybe somehow they are related to Xwayland.

To be honest I'm not a fan of Wayland because I experienced so much problems with it. For me it feels like Wayland is either far away from being mature, or their security "features" are way too restrictive. For example to create screenshots and screencasts for the LibrePCB documentation/website, I still cannot find a reliably working capturing tool after so many years of the existence of Wayland. The developers of these tool say that Wayland is a nightmare for them, not providing the proper APIs etc. while on X11 everything is available and (from user perspective) works very well.

doom-fr, 2023-08-29:
Thanks for your help !

closed 2023-08-29 by doom-fr