LibrePCB/LibrePCB #1231: Potential Library editor/manager improvements

hooperbloob, 2023-09-21:
re: v1.0.0 rc1

As I click on the entries across the six lists shown it would be helpful if some additional details
for the selected item could be shown on the right hand panel.  At the moment that panel is fixed, 
showing details of the overall library, this is wasted space for the most part and should be positioned 
on the left side anyways as it semantically encompasses everything.

For example if I click on Capacitors in the ComponentCategories list I'd like to see it auto-select any 
related symbols selected in the next column.. and then show the symbols on the right along with any 
associated packages they might reference. You could put a checkbox above the dependent lists to 
enable/disable that selection behaviour.

I think each of the lists should also include counts of each item:

Capacitors          34
Comparators         4
Connectors         12

If the item creation date is captured it'd be nice to provide a date filter so that new components could be
highlighted.

Please verify first that your request is not already reported on GitHub.

Attention: This issue tracker is only for "smaller" feature requests (e.g. "Make
layer colors configurable") or usability flaws (e.g. "Pressing Enter does not
close the dialog"). For fundamental, complex, large features (e.g. "Implement 3D
board viewer") please use our discussion platform instead so we can first
discuss how the feature should be implemented exactly:

https://librepcb.discourse.group/t/about-the-improvement-ideas-category/12
-->

##### SUMMARY

##### SOLUTION

ubruhin, 2025-08-03:
Almost all of the suggested features are implemented in #1553. The only one left is this:

> If the item creation date is captured it'd be nice to provide a date filter so that new components could be
highlighted.

I really like that idea. I'd suggest we just highlight all elements which were created in the past 24 hours - no manual filter feature, no configurable time span etc, just put some "new" indicator beside all new library elements.

As this is the only remaining feature request of this issue, I'll adjust the issue title accordingly.

hooperbloob, 2025-08-04:
I'm happy that you like the idea, I'd suggest an alternate means of noting the more recent ones than using highlighting.  

If you add a new column with coloured dots, you could represent their age with colour. I.e. new ones are red hot, older ones are dark blue and everything in between. They would show the age distribution of the parts in the library in a meaningful way. Clicking on the header would sort by date also.  Hovering over the dots would reveal the date and possibly other metadata.

Heck, you could use different shapes to denote other aspects:  whether they are used in current projects, how popular they might be, their possible validation status..etc.