LibrePCB/LibrePCB #1234: FileUtils::getFilesInDirectory reveal nested hidden files

BuFran, 2023-09-25:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 1.0.1-unstable
Git Revision: 8d5a2b1b
Qt Version: 5.15.3
Operating System: Linux Mint 21
...
```

##### SUMMARY
I created UnitTests for FileUtils and this function behaves incorrectly. This describes first observed bug.

According to doc:

```
@brief Get all files in a given directory (optionally filtered by
   * extension)
   *
   * @param dir             Filepath to a directory (must exist)
   * @param filters         Only files matching this filters are returned
   * @param recursive       If true, also subdirectories are searched for files.
   * @param skipHiddenFiles If true, hidden files won't be returned.
   *
```
the function should find all files in all subdirectories when `recursive=true`, and filter out hidden files when `skipHiddenFiles=true`.

##### STEPS TO REPRODUCE
I created test case for testFindFileInDirectoryRecursiveSkipHidden in PR #1233

The structure of directories inside test is:

```
 /file.txt
 /.hidden.txt
 /subdir/file.txt
 /subdir/subdir/.hidden.txt
 /subdir/subdir/file.txt
```

##### EXPECTED RESULTS
```
  /tmp/librepcb/1695651212481_1882556969/file.txt
  /tmp/librepcb/1695651212481_1882556969/subdir/file.txt
  /tmp/librepcb/1695651212481_1882556969/subdir/subdir/file.txt
```

##### ACTUAL RESULTS
```
  /tmp/librepcb/1695651212481_1882556969/file.txt
  /tmp/librepcb/1695651212481_1882556969/subdir/file.txt
  /tmp/librepcb/1695651212481_1882556969/subdir/subdir/.hidden.txt
  /tmp/librepcb/1695651212481_1882556969/subdir/subdir/file.txt
```

##### ADDITIONAL INFORMATION
The test revealed that recursive call doesn't manage all arguments in recursive call, so nested hidden files are reported wrongly and are not filtered out (skipped).

PR correcting the bug will follow shortly

ubruhin, 2023-09-25:
Huh, nice catch! :+1: Will check the PR soon!

ubruhin, 2023-09-27:
Fixed in #1235

closed 2023-09-27 by ubruhin