LibrePCB/LibrePCB #1236: FileUtils::getFilesInDirectory is not well documented, has surprising behavior

BuFran, 2023-09-25:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 1.0.1-unstable
Git Revision: 8d5a2b1b
```

##### SUMMARY
The function header is:
```
  /**
   * @brief Get all files in a given directory (optionally filtered by
   * extension)
   *
   * @param dir             Filepath to a directory (must exist)
   * @param filters         Only files matching this filters are returned
   * @param recursive       If true, also subdirectories are searched for files.
   * @param skipHiddenFiles If true, hidden files won't be returned.
   *
   * @return A list of filepaths to files in the specified directory
   */
  static QList<FilePath> getFilesInDirectory(
      const FilePath& dir, const QStringList& filters = QStringList(),
      bool recursive = false, bool skipHiddenFiles = false);
```
The actual work inside function when `filter` is specified and `recursive` flag is set is different than obvious by name and docstring.

##### STEPS TO REPRODUCE
none needed, #1233 

##### EXPECTED RESULTS
when searching for files recursively, function should return **all files matching pattern** from **all subdirectories** starting at root.

##### ACTUAL RESULTS
function returns **only** files matching pattern in subdirectories where subdirectory name **matches pattern**

##### ADDITIONAL INFORMATION
It can be corrected two ways.

 * One way is to document this unintended behavior in docstring.
 * Second way (I prefer) search for files and directories in two loops inside as this is native understanding of functionality.

If the first version will be selected, #1233 should be merged fully, If the second, #1233 last commit should be dropped before merging.

ubruhin, 2023-09-27:
Good point, luckily this didn't cause any problem yet since name filter and recursive are never used at the same time (yet) :sweat_smile: 

I agree your proposed second way is better :+1: Probably it makes sense to find files first, and then within `if (recursive) {}` find subdirectories. The checks for `info.isFile()` and `info.isDir()` are then no longer needed anyway, so we get two simple loops instead of one loop with conditionals.

So I'm happy to receive a PR for that :) I'd then wait with #1233 for now so it can be updated to the new behavior before merging.

ubruhin, 2023-10-02:
Fixed in #1241. Thanks!

closed 2023-10-02 by ubruhin