LibrePCB/LibrePCB #1240: Document how to test newly localized strings from Transifex

BuFran, 2023-09-27:
##### SUMMARY
There is some info on the pages informing that volunteers can help by translating parts of software on Transifex. 

But there is missing description how to test the app with new (edited) strings to check if translations are correct.

##### SOLUTION
User should download codebase and make it compilable to get tools like `lupdate` working.

Before build on actual `master`, volunteer can replace file `i18n/librepcb_XX.ts` with newly downloaded translation from Transifex.

After build on actual `master`, there will be newly compiled `i18n/librepcb_XX.qm` in the build directory.

But after run application, only "Default language" in the settings is shown, no way to select any translated language. Only installed version of app shows languages built during production release.

There should be some step-by-step description how to make languages available to custom-built application. Seems they are searched in different directory than they was built.

BuFran, 2023-10-03:
I looked into workspace settings dialog code and the compiled `.qm` needs to be copied from the `<git root>/cmake-build-debug/i18n` folder to `<git root>/share/librepcb/i18n` to be used as translation source files. 

In some circumstances the folder may be different, look at debug output from console and use this destination, under `i18n` subdirectory:

```
[  INFO   ] Resources directory: "<somewhere>/LibrePCB/share/librepcb" (<somewhere>/LibrePCB/apps/librepcb/main.cpp:150)
```

After this move, the translations are ready to use for review of any language.

Remember to **delete** them after review/test, because that this folder is source folder under git control and should not contain binary data.

Am i correct ?

ubruhin, 2023-10-03:
Generally your described way works. But for our official builds, translations are installed to the correct location with `make install`:

    mkdir build && cd build
    cmake .. -DCMAKE_INSTALL_PREFIX=/path/to/librepcb-output
    make -j8
    make install

Afterwards the binary is located at `/path/to/librepcb-output/bin/librepcb` with working translations.

BuFran, 2023-10-05:
I was looked on this issue to resolve it and seems this paragraph should be at the end of 

[LibrePCB/CONTRIBUTING.md](https://github.com/LibrePCB/LibrePCB/tree/master/CONTRIBUTING.md) ?

Your version with `INSTALL_PREFIX` looks much better!

I'll try to make PR for that documentation change to close this issue.

ubruhin, 2023-10-05:
Note that `CONTRIBUTING.md` is just a very rough overview about the whole code contribution workflow. The details (including the `INSTALL_PREFIX`) are documented at https://developers.librepcb.org, (CMake flags [here](https://developers.librepcb.org/d5/d96/doc_building.html)).

Or in other words, `CONTRIBUTING.md` is for *code contributions*, not translation contributions (since they are not PRs to the source repository). For translation contribution guidelines I'd propose to use a separate location since it is a totally different thing. I'm not sure yet where the best location would be for that (either a separate page on https://developers.librepcb.org, or maybe a post in the forum since this could entirely be organized by translators themselves, without requiring my approvals like on the source repository).

Or ideally such documentation would be placed directly at Transifex, but I don't know if this is somehow possible(?).