LibrePCB/LibrePCB #1246: Crash at launch with segmentation fault

julian-weinert, 2023-10-03:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 1.0.0
Operating System: macOS 10.14.6 (18G9323)
...
```

##### SUMMARY
The current downloadable binary of the 1.0.0 release does not launch on macOS. After retrying a few times I executed the binary directly from the console. That revealed that it suffers a segmentation fault immediately.

##### ADDITIONAL INFORMATION
I cloned the repository and compiled the 1.0.0 branch (Qt 5.15.7). In that case it launches and runs as expected.

ubruhin, 2023-10-03:
Thanks for the bug report. Unfortunately I have no way to test/debug the binary on macOS 10.14. Maybe the macOS version on Azure CI is too new, producing binaries not compatible with such old macOS versions anymore. But we have no other option, Microsoft doesn't provide older macOS runners.

So my only suggestions are:
- Try the LibrePCB Homebrew Cask, maybe that works: https://formulae.brew.sh/cask/librepcb
- Continue using the self-built version :wink:
- Upgrade macOS. Sorry I hate such recommendations, but in case of macOS the big tech companies enforce us to use the latest versions, otherwise they punish us with pain :sob:

julian-weinert, 2023-10-09:
I built it myself now and it runs without complaints...
Anything I could do to figure out what happened to the public build?

ubruhin, 2023-10-09:
>  Anything I could do to figure out what happened to the public build?

I don't think so... Actually since the self-built binary runs fine, I'm pretty sure it's some binary incompatibility between macOS 10.14 and the bundle built on macOS 11. Although we explicitly specify in CMake to target macOS 10.14 (for maximum compatibility), our dependencies might not do that so those libraries might fail to load.

Since I am not familiar with macOS, may I ask what's actually the reason for not upgrading macOS to a later version? According [Wikipedia](https://en.wikipedia.org/wiki/MacOS_version_history) macOS 10.14 is not supported anymore.

closed 2023-11-22 by ubruhin