LibrePCB/LibrePCB #1250: Run possibly unstable operations in a separate process

ubruhin, 2023-10-04:
##### SUMMARY

I do not want to blame the OpenCascade library (it is an impressive piece of software), but this library (either the library itself, or maybe our usage of the library) already lead to tons of crashes of the whole application. This leads to very bad user experience, and possibly causing people to blame LibrePCB for issues we're not responsible for. Beside the OpenCascade integration, there are also other pieces of (complex) code which are a risk for the stability of LibrePCB.

Please don't read this wrong. Overall, I think the stability of LibrePCB is pretty good (at least compared to other EDA tools :joy:). But I have high expectations on stability and thus I'm sure there's room for improvements.

##### SOLUTION

Crashes in complex code usually mean *segfaults*. So at least in third-party libraries we can not do anything on our side to avoid these crashes. Therefore I wonder if it would make sense to move "risky" operations into a separate process, to avoid crashing the whole LibrePCB application if something goes wrong in that operation. Of course only operations which are rather isolated (and not performance critical) can be extracted. For example reading STEP files and returning the triangles for rendering on screen would be a perfect candidate.

Unfortunately, starting a new process is pretty slow on Windows. So I'm not sure if it's a good idea to move every risky operation into a separate process which gets started on demand. Probably the performance would be much better if there's a single "worker process" which is started only once and kept alive during the whole lifetime of the LibrePCB application. When it crashes, it should simply be restarted. Operations should be passed as "jobs" to the running process, which could for example start each job in a separate thread.

This is just a rough idea. There are still several things to be evaluated:
- How are jobs (with input/output data) passed to the worker process (IPC)?
- Which operations can/should we move to a separate process? Ideas:
  - Everything which uses OpenCascade, i.e. reading & writing STEP files
  - Background workspace library scanner?
- How is the separate process deployed, resp. does starting and communicating with it work fine even on sandboxed environments like Snap/Flatpak?

BuFran, 2023-10-05:
Stick different solution to be considered (without threads, just catch exceptions):

[segvcatch](https://github.com/Plaristote/segvcatch)

BuFran, 2023-10-06:
(work in progress comment, please please do not answer/comment yet, i will edit it many times as I will develop solution through time, it will contain wrong assumptions until considerations be made and possible variants of solutions will be proposed)

This looked simple at the beginning but after consideration, this is huge problem.

## Threads/Processes

Initially I mean that this can be used using posix threads and this can enable making "worker threads" they can speed-up some operations (load library for example) in the near future on background. Threads can be simply destroyable. 

But this is not true because we can't catch segv same way as Java or other garbage-collected system (as stated above) and the process causing segv should terminate as fast as possible. It should be implemented only with separate processes, not threads.

## Design considerations at the edges

* Worker should receive data
* Worker should give result back
* Worker should give feedback about it's progress
* Worker must be able to die (fail)
* Sender should be able to enqueue work for worker
* Sender should cancellable wait for result
* Sender should detect if worker died and restart it.
* Sender should be able to show progress of operation

dbrgn, 2023-10-08:
@ubruhin I like your suggestion. Note: Workers and message passing and restarts on failure sounds a lot like the "actor model". Maybe there are some nice C++ libraries that implement/abstract the actor model in a cross platform way?

ubruhin, 2023-10-08:
> Note: Workers and message passing and restarts on failure sounds a lot like the "actor model". Maybe there are some nice C++ libraries that implement/abstract the actor model in a cross platform way?

I think there exist such libraries but as far as I can see they are huge and complex and way too heavy for our simple use-case (we don't want to build a scalable, distributed worker farm or something like that). Actually I really hope this can be implemented without requiring _one more_ third-party library :see_no_evil: 

Qt already provides the most important classes to start processes and communicate with them in a portable way, see https://doc.qt.io/qt-5/ipc.html. At least with `QProcess` (which we already use for other things) it should be easy to start a worker process (with automatic restart on failure). But of course Qt doesn't provide a complete solution how to pass our data structures between the processes.

BuFran, 2023-10-16:
The main issue here from my opinion is the serialization/deserialization of data, that can be pretty slow. 

All other things can be solved multiple, simple ways (see my comment above - still editing it to fit use case using simplest objects they already in Qt5). 

So I recommend to use it only for operations containing *small data* or *data already in files*. 

It can therefore benefit in case of "merging STEP files", but not "loading them" because it involves large layer of serialization and deserialization where it is bigger possibility to *insert new bug* instead of *suppress alien bug*.

Sandboxing is second big issue here.

BuFran, 2023-10-16:
> Well, but then the whole idea of this issue is gone.

Yeah, I see that in same way. When you have object in memory, it contains pointers to process-local page and you should not expose it's binary blob directly to other process as this process can't access it. There should be always some "interconnect layer" in the architecture, to which we can/should serialize from both sides. 

As I looked in the code, most of the objects we discussing about them, they have serialization operators `<<` and `>>` implemented so the serialization/deserialization work is done now. However, as You stated, those operators contains operations we like to check/sanitize so they can't be used so easily and the whole idea is moving to recurrent infinity.

So for now, should I dig deeper into **streaming** or **network** solution (as you recommended me via Telegram chat) or we can close this issue for now?

ubruhin, 2023-10-16:
> As I looked in the code, most of the objects we discussing about them, they have serialization operators << and >> implemented so the serialization/deserialization work is done now.

No, these operators are not meant for this kind of serialization, only for logging. Serialization for IPC needs to be implemented separately.

> So for now, should I dig deeper into streaming or network solution (as you recommended me via Telegram chat) or we can close this issue for now?

I'd keep the issue open since I still think it makes sense to implement. Feel free to dig deeper into streaming and network solution. However, as I wrote initially, I'm not convinced this is a good issue for new contributors since in the end it is quite difficult (not that complex technically, but involving many aspects - concept design, software architecture, implementation, testing on various systems, deployment/packaging [snap/flatpak sandboxes!], requiring lots of knowledge about the current software architecture). So I'd also be fine to keep your current investigation for later (thanks so far!) but then continue with different issues. It's up to you.