LibrePCB/LibrePCB #1252: Restructure code to utilize a plugin system

GravisZro, 2023-10-05:
##### SUMMARY
LibrePCB is getting quite large which means it's become difficult for others to make contributions on the sole basis that they do not know the entire codebase like you do.

##### SOLUTION
A plugin system would enable people to work on parts of the project without having to understand the entire project. It would also simplify the build system options (way too many options IMO) by enabling people to either build the plugin and enable it at runtime or not.

If you want this to become more than a one-man show then the code will need to be segmented into individual components instead of a single massive codebase.  If decide against this then LibrePCB both lives and dies with you.

ubruhin, 2023-10-05:
Note that this idea already exists for many years: https://librepcb.discourse.group/t/add-ability-to-load-plugins-at-runtime/32

Generally I absolutely agree plugins would be useful, but the big questions are:

1. What problem do they intend to solve?
2. How are they technically implemented (and deployed)? This is a quite complex topic, especially if plugins are written in a compiled language like C++ since binary compatibility of them with the application becomes a big problem.

Since your issue seems to target question 1.:

> If you want this to become more than a one-man show then the code will need to be segmented into individual components instead of a single massive codebase. If decide against this then LibrePCB both lives and dies with you.

This sounds great, but to be honest I do not think *any* software architecture would change this. I also had this "romantic" vision of how open-source projects work. But now after working ~10 years in the open-source world, I had to correct my vision. In the end, it's not the software architecture or the number of contributors which make a project resilient (in terms of "lives and dies with [a single person]"). Basically *any* software which is developed by volunteers has this problem. One reason is for sure that contributors underestimate the effort required to keep such a project alive and are not willing to spend this amount of free time for it. Another problem is simply that for such a project there are tons of credentials needed (domain purchasing, email account, server login, CI management login, and many many more) which a project leader cannot make just public (also not to contributors you don't know very well). So if the project leader dies, nobody has access to all these things.

Thus, my corrected vision is now that even an open-source project can only be operated in a resilient way with (sorry) money. With money, a project can be driven by a few, very well connected, everyday working & communicating developers which trust each other. This way you can more easily share knowledge and credentials about the project. That's not possible with 1000 random contributors changing every day. With working on a project as a full-time job, developers won't stop working on it just because they started a new hobby or found a girlfriend. So the project is much more resilient than when operated entirely by volunteers.

I know this is an unpopular view (my open-source heart bleeds to say that) but this is what I learned as a project leader for many years. If you know a big, complex open-source project which is operated in a resilient way without lots of money, please let me know :wink: The KiCad project also only got popular and more professional after they received a significant amount of money (if I remember correctly, their budget is now >200k USD per year).

Long story short: The problem of the "one-man show" cannot be solved by any software architecture. Btw. I think the LibrePCB [architecture](https://developers.librepcb.org/de/d34/doc_architecture_overview.html) is actually not that bad, especially when compared to other software projects :see_no_evil:

BuFran, 2023-10-06:
Hi gravisZro, I like your issue, and if I can vote for that, i will give vote +100 for that. So let's look at the problem from other point I see here. 

U.Bruhin spent with this software a lot of time. If you look at Git history, there has been a lot of people contributing at the beginning of the project. After time, there is only him, who pushed some commits so the project style has been moved from "chaotic evolution" to "one man directed (dictated) show". From that point of view, you can see that he *should be* afraid that some day someone comes, without knowledge of internals, he proposes some feature which doesn´t fit into the model, partially implements (not worrying about side effects) and leaves the team without catching those problems. And he, the maintainer will be the person who needs to manage and fix the problems the other man did. Well in that way we must accept, that this project slowed down, and only things that "one man" can manage can be implemented.

If you look at code (and i spend a lot of time in past week(s) to analyzing his architecture), the code is very clean, understandable, there is some slight misconceptions but the architecture is solid from top overview. It is the result of "one man show" where the one can design and maintain everything. It is signature of U.Bruhin, and we should agree, that if we (contributors) add something new, what disturbs the result, it will be nothing good for U.Bruhin's signature (app will crash after contribution of someone, but people will yell on U.Bruhin). So *we should agree* with bruhin's controlled contribution under authorization.

From that view points, **this issue** looks completely different. By implementing some *plugin api* or *scripting*, everyone on that planet can build every nice functionality, share it and connect with his work and someone other can yell at him "it is not working when XX and YY" and the project will be unmanageable then by one person, because it will be out of control by his person. That *is* in my mind the exact reason why this was not implemented for years and we *should accept* it. It's life.

The complete different thing is, that there is no straight, visible "future plan" of the project, only a unorderable set of issues that user people wants, that contributors are *forbidden to implement or fix* because "there is long term refactoring of UI planned" or "this is misconception in model, but we should mark it for larger refactoring in future" and even if someone wants to contribute and solve some minor issues, they are still open for years, so many contributors surrender the contribution during (or before) review process and loop starts at the beginning, at one man, maintainer who need to implement everything, without help of others. And issue list is constantly growing :-( 

So the real questions about every issue here is: 

 * Are we surely need this feature / can we live with this bug ? 
 * When we need this feature implemented, it is "must have soon" or "well it doesn't matter for now"
 * How much effort it takes ?
 * Who will design the solution (architecture) ?
 * Who will contribute on this feature, who solve it ? He needs to know internal architecture!
 * Who will test this feature, it will not contain new bugs ? On all supported platforms ?
 * Are someone willing to pay for that feature ?

If you answer U.Bruhin on more points above, unfortunately there will be result in "one more issue remaining here we should live with". 

According to this issue, my opinion (as young contributor in this repo) is:

 * no, for primary functionality of this app, this issue is not needed
 * maybe in future, its nice but there are other problems to solve
 * very
 * Only maintainer can do this :-(
 * noone known for now
 * Only maintainer can do this :-(
 * GravisZro ? :-) 

This text is my personal opinion of one man who have time until end year to contribute and trying to fix something here. Next year ? Who knows (*this is exactly what maintainer said in previous answer and he is afraid!*)

BuFran, 2023-10-06:
To have some addition to this issue. When you look at codebase in git, there was clean distinction of core, user interaction and application (cli/gui). 

Somewhere in time, the split has been merged back into one big library containing entire app with user interaction and two applications (cli and gui).  So only three parts of app are visible from outside. This merge is reasonable because only maintainer managed this project during that time.

To solve this issue, the data core (model) should reside in separate library, with no dependency on gui, forms etc. This enables linking of core to the plugin, and it is "required" for any plugin system to work.

ubruhin, 2023-10-06:
Generally I agree on the points mentioned by @BuFran - thanks for understanding my point of view as the project leader :slightly_smiling_face: 

Just a small comment:

> Somewhere in time, the split has been merged back into one big library containing entire app with user interaction and two applications (cli and gui).
> [...]
> To solve this issue, the data core (model) should reside in separate library, with no dependency on gui, forms etc.

Here I don't agree :see_no_evil: There are two clearly separate libraries (not just directories, see CMake setup) `core` (the data model) and `editor` (all the GUI stuff) while the core library has no dependency to GUI at all. Although there were more libraries in the past (see #939), I think it does not mean there are too few libraries now. The separation of common/library/project/workspace is not really important, all of them are representing the data model so one library for them is actually enough (and fixes several problems we had before).

I think the CLI is a good example that the architecture is actually very clean. Just [a lightweight layer](https://github.com/LibrePCB/LibrePCB/blob/master/apps/librepcb-cli/commandlineinterface.cpp) on top of the `core` library is enough to implement a perfectly fine working CLI, including nice error handling etc. No code duplication, no dependency to GUI, no CLI code outside the CLI application.

Anyway, as I already stated in my first post, I am not against a plugin interface, actually I do think it is a good idea if it is well designed.