LibrePCB/LibrePCB #1253: UI Rework suggestion

GravisZro, 2023-10-06:
##### SUMMARY
Since the UI is apparently being reworked, I suggest a design based on the Qt Creator GUI design.  The mock up doesn't cover everything about the GUI but I think it's a solid basis for turning the multiple windows GUI into a single window GUI.

[image]

ubruhin, 2023-10-06:
Thanks for this suggestion! Interestingly your mockup looks basically identical to mine :smiley: 

[image]

GravisZro, 2023-10-07:
Well... it looks like my work here is done.

closed 2023-10-07 by GravisZro