LibrePCB/LibrePCB #1255: Installer Fails at 99% - Reports downloaded source file is corrupt...

sproggit, 2023-10-10:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 1.0.0
Git Revision: 1.0.0
Qt Version: No Idea
Operating System: Windows 10/64-bit/Pro
...
```

##### SUMMARY

Download librepcb-installer-1.0.0-windows-x86.exe
Check and confirm that Windows Defender finds no issues
File downloads at 21,565,846 bytes
Did not see a checksum on the download page? 

##### STEPS TO REPRODUCE

Download file - attempt to install
After fail, delete file, re-download and re-attempt
In both cases, the installation proceeds to 99% and then reports a fault with the source file before backing out the installation.
On both occasions, I let the installer run with default installation parameters

##### EXPECTED RESULTS

Expected the installer to finish and the application to be installed 

##### ACTUAL RESULTS

Got to 99% and then saw the reported error

The installer creates a pop-up which reads, "Cannot open the file {point to my downloaded copy of the installer} for reading: Unknown error" 

If I select the "Show Details" Option, I get the following:-

...
Installing component Register File Extensions...
Extracting "0.2.0registerfileextensions.7z"
Done

Installing component Common Files...
Done

Installing component LibrePCB Applicaiton...
Extracting "1.0.0-1stable.7z"
Done
...

And then I get the error message. 

##### ADDITIONAL INFORMATION

NOT, NOT, NOT a major priority... 
I am happy to experiment, check system config more broadly, etc. 
Am going to try a Linux install to see if I can make that work.

ubruhin, 2023-10-10:
I can not reproduce the problem, I think it must be related to your local setup. I found some hints that it might be related to network drives, please make sure that both the downloaded installer and the installation directory are located on a local drive. Also make sure there's enough free disk space. If it still doesn't work, try to disable antivirus if any (they are causing so many random problems).

>  Am going to try a Linux install to see if I can make that work.

You could also try the portable *.zip for Windows which is provided on the download page.

>  Did not see a checksum on the download page?

https://download.librepcb.org/releases/1.0.0/sha256sums.txt

sproggit, 2023-10-14:
Many thanks for the prompt response. 

Checksum confirmed a good download file... I checked and I had inadvertently saved the downloaded file directly to my NAS and I was attempting to install from there. Relocated the installer to a local drive and it ran without any problems. 

_Hangs head in shame._ 

Sorry to have troubled you.

closed 2023-10-14 by sproggit

sproggit, 2023-10-14:
USD $25 Donation via PayPal in appreciation for both the software and your assistance. Thank you.

ubruhin, 2023-10-14:
No worries, good to know it's not an issue on our side ;-)

Donation received, thank you very much! :heart: