LibrePCB/LibrePCB #1257: Install Libraries

brodao2, 2023-10-13:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 1.0.0 build date 2023-09-24 08:07:32 (Hora oficial do Brasil)
Git Revision: 9edb6ede3
Qt Version: 5.5.2
Operating System: Windows 11 22H2
...
```

##### SUMMARY

I click on "library manager" and the recommended ones are already marked.
I click "download and install..." and a progress bar appears and all completions are displayed without an error indication.
Reopen the "library manager" and the recommended libraries are not installed.
Intermittently, I can install one or two of them if I select them one by one.
On the desktop, the downloaded library folders appear.

##### STEPS TO REPRODUCE

1. Click "Library Manager"
2. Click "Download and install..."

Or

1. Click "Library Manager"
2. Mark "Select All"
3. Click "Download and install..."

##### EXPECTED RESULTS

Installed libraries.

##### ACTUAL RESULTS

Libraries are downloaded but not recognized as installed

##### ADDITIONAL INFORMATION

[image]
[image]
[image]
[image]

ubruhin, 2023-10-14:
Hm, I never saw this problem. Would be good to have some more information to understand why this happens:

- Did you wait long enough for the installation to complete (unpacking ZIP is very slow on Windows)?
- Enough free disk space and RAM available? Maybe watch at the task manager during the library installation.
- Is your workspace located on a normal, local disk (not on a network drive)?
- Any reason for using Qt 5.5.2, resp. could you try with a more recent version?

korpa, 2023-10-31:
Same issue here. I use the windows portable app. Free disk space is ok. RAM is ok. Workspace is on a local disk, but on the second partition (D:).
The file are installed correctly on the disk, but it seems, that the library scanning does not work:

I start with one library installed.

[image]

After selecting the second library and clicking on download & install, the ZIP is downloaded and extracted. But in the main window no "scanning library" appears in the right bottom.
Furthermore the library manager understands, that the lib is on disk and removes the checkbox. But it doen't appear on the left side. 

[image]

When I click on "Rescan libraries" in the menu of the main window, nothing happens.

ubruhin, 2023-10-31:
Thanks for the details @korpa!

I think it would be very helpful to see the logging output of LibrePCB while this problem occurs, since any problems of the rescan should be logged. Could you maybe provide us such a log? Here the steps to create it:

1. Run LibrePCB from the Windows command prompt with this command:
    ```
    C:\path\to\librepcb.exe > log.txt 2>&1
    ```
2. Install libraries to trigger the described problem.
3. Manually trigger "Rescan libraries" and wait until it's finished (progressbar at the bottom).
4. Close LibrePCB.
5. Post the content of the created logfile `log.txt` here (feel free to obfuscate your username in the filepaths).

korpa, 2023-10-31:
That's really strange. When starting it from within a Powershell with stdout and stderr redirects like you asked me to do, download will stay at 0% forever:

[image]

korpa, 2023-10-31:
If I start it from Powershell without the redirects, behavior is like if I start LibrePCB via clicking librepcb.exe within explorer:

[image]

ubruhin, 2023-10-31:
Hmm the command I posted is intended for `cmd.exe`, I don't know if this is also valid for PowerShell. In my experience, PowerShell has very strange behavior so I'm not surprised anymore if it doesn't work as expected :see_no_evil: Could you thus also quickly try it from `cmd.exe`?

korpa, 2023-10-31:
`cmd.exe` worked.

Here you go
[log.txt](https://github.com/LibrePCB/LibrePCB/files/13216324/log.txt)

ubruhin, 2023-10-31:
Thanks! This log is indeed very interesting. Could you test if the following steps solve the problem?

- Make sure LibrePCB is closed.
- In your workspace subdirectory `data/libraries/` delete all files starting with `cache_`.
- Open LibrePCB again and try installing libraries.

korpa, 2023-10-31:
Yes, this works. Everytime I close LibrePCB, delete `cache_v5.sqlite` and restart LibrePCB again, the lib I installed before closing LibrePCB, appears in the list. Did this twice and now it looks like this:

[image]

So at least I have workaround for now.

ubruhin, 2023-10-31:
Ok good to know. But do I understand correctly that the problem occurs again when installing more libraries, i.e. the workaround helps only temporarily?

If the problem still exists, I'd highly appreciate if you could generate the same logfile again, but this time with [this nightly build](https://download.librepcb.org/nightly_builds/develop/librepcb-nightly-windows-x86.zip). It produces some more logging messages to help me finding the root cause. That would be really great. Note that this time it would be good to keep the original filepaths in the logfile since they could be relevant. You could send the file to our [contact email address](https://librepcb.org/contact/) and I will handle it confidentially.