LibrePCB/LibrePCB #1258: Support trackpad gestures for zoom/pan

ubruhin, 2023-10-14:
##### SUMMARY

Pinch to zoom has been implemented for macOS a long time ago: https://github.com/LibrePCB/LibrePCB/pull/477

But it seems this doesn't work on other operating systems. See also this forum thread: https://librepcb.discourse.group/t/pan-and-zooming/590

##### SOLUTION

- Pinch on the trackpad shall zoom instead of scrolling.
- Dragging with two fingers shall pan instead of zooming.

I'm not sure if there is any way to implement that without breaking the mouse navigation. I realized other Qt users have the same problem and didn't find a solution. However, it's something quite important for usability so this should be investigated.

p12tic, 2024-01-20:
On Linux this should just work on new enough Qt and Xorg/Wayland.

Tubbles, 2025-03-13:
Can confirm this is still an issue for me with wayland. two-finger pinch on the trackpad does nothing while two-finger dragging results in zoom

`Linux 6.12.10-100.fc40.x86_64`
`Fedora release 40 (Forty)`
on flatpak librepcb version 1.2.0

ubruhin, 2025-05-24:
Update: As we are now using Slint, this feature depends on that one: https://github.com/slint-ui/slint/issues/6704