LibrePCB/LibrePCB #1261: No solder mask

eto-, 2023-10-29:
##### SUMMARY

I would like to have regions of a copper plane not protected by the solder mask.
I was not able to find a way that is not remove all the solder mask form that surface. 

##### SOLUTION

ubruhin, 2023-10-29:
With the "Draw Polygon" tool on the "Top/Bottom Stop Mask" layers you can add custom solder mask openings. Or are you asking for automatically opening the whole copper area of a plane? If so, it would be interesting to see a practical example to decide whether this makes sense to implement. At least I never saw such a feature in other tools.

eto-, 2023-10-29:
My fault I did not realize the default line was size 0 and therefore I could not see it in the 3D.

Thanks

closed 2023-10-29 by dbrgn