LibrePCB/LibrePCB #1264: Few small improvements

eto-, 2023-11-02:
##### SUMMARY

The default size for the Names (top & bottom) is 1 mm which is way too big for SMD devices below 1206 (Imperial). I did not find a configuration where to set it (at board or main level). The result is that after placing all 100+ components I have to manually modify the size.
Similarly is it possible to permanently disable the Values? I normally do not see them in modern PCBs.

Sometime it happens that I have "empty fragment" in the DRC: I can edit the file to remove the corresponding line, but maybe a "fix" button could be easier for most users.

##### SOLUTION

ubruhin, 2023-11-02:
> The default size for the Names (top & bottom) is 1 mm which is way too big for SMD devices below 1206 (Imperial).

That's true, but what would be your suggestion what we should change? We don't use text sizes smaller than 1mm because they might not be readable on silkscreen anymore. AFAIK most manufacturers specify a minimum text height around 0.8-1mm. I suspect for tiny SMD devices you don't put the name on silkscreen at all, or how do you handle that?

>  Similarly is it possible to permanently disable the Values? I normally do not see them in modern PCBs.

Currently not but I agree values are not that important, that's also the reason why they are not put on silkscreen by default. However, they could be useful for creating assembly plans for printout. Generally if you're not interested in values at all you can just hide the value layers (but I agree it might be handy to configure that on board level).

> Sometime it happens that I have "empty fragment" in the DRC: I can edit the file to remove the corresponding line, but maybe a "fix" button could be easier for most users.

Huh that's strange, this DRC warning is just a safety net to detect such fragments, but actually they should never appear at all. If you ever find out how to reproduce them, please let me know the exact steps. Anyway, a "fix" option would be nice :+1:

Btw, it would be better to create separate issues for unrelated topics as it helps us to prioritize and work on them individually ;)

eto-, 2023-11-02:
> That's true, but what would be your suggestion what we should change? We don't use text sizes smaller than 1mm because they might not be readable on silkscreen anymore. AFAIK most manufacturers [...]
Actually for the producers I know, the minimum width of a font line is 6 mills (150 um): this correspond to about .6 mm height for the fonts present on LibrePCB (I measured with the tool manually). Therefore I typically use 0.7 mm that corresponds to 200 um font lines.

> 
> > Similarly is it possible to permanently disable the Values? I normally do not see them in modern PCBs.
> 
> Currently not but I agree values are not that important, that's also the reason why they are not put on silkscreen by default. However, they could be useful for creating assembly plans for printout. Generally if you're not interested in values at all you can just hide the value layers (but I agree it might be handy to configure that at board level).
I typically disable the value layers but then when I switch to another configuration (top to bottom) they get enabled back.

> Huh that's strange, this DRC warning is just a safety net to detect such fragments, but actually they should never appear at all. If you ever find out how to reproduce them, please let me know the [...]
I will experiment on this.

> 
> Btw, it would be better to create separate issues for unrelated topics as it helps us to prioritize and work on them individually ;)
Roger that.

ubruhin, 2023-11-02:
> Actually for the producers I know, the minimum width of a font line is 6 mills (150 um): this correspond to about .6 mm height for the fonts present on LibrePCB (I measured with the tool manually). Therefore I typically use 0.7 mm that corresponds to 200 um font lines.

Yeah some manufacturers are able to produce that, but for example [JLCPCB specifies](https://jlcpcb.com/capabilities/Capabilities) a minimum text height of 1.0mm. Thus I think for libraries it's much safer to stick with a minimum of 1.0mm. An alternative might be to allow smaller texts in the library and automatically enlarge them depending on a board setting, but I'm not sure if this is worth the additional complexity... Probably a multi-selection tool would be good enough, i.e. allowing to select multiple texts and changing them from 1.0mm to e.g. 0.7mm all at once.

> I typically disable the value layers but then when I switch to another configuration (top to bottom) they get enabled back.

Yeah that's true. Would be nice to have a clean way to handle that...

eto-, 2023-11-03:
> Yeah some manufacturers are able to produce that, but for example [JLCPCB specifies](https://jlcpcb.com/capabilities/Capabilities) a minimum text height of 1.0mm. Thus I think for libraries it's [...]

That would be great: in general I miss the group action of Eagle (being able to change in bulk the size of vias, the net size and so on).

closed 2023-11-05 by eto-