LibrePCB/LibrePCB #1265: Does windows installer support silent update?

luzpaz, 2023-11-06:
##### SUMMARY

Trying to get LibrePCB into downstream winget (windows) package repository. The question was brought up by a maintainer 

See https://github.com/microsoft/winget-pkgs/issues/124364#issuecomment-1794350224

##### SOLUTION

n/a

ubruhin, 2023-11-08:
Hi,

> Trying to get LibrePCB into downstream winget (windows) package repository.

Awesome, thanks for working on this!

I'm not sure if this is what you're looking for, but I think there's a `--silentUpdate` CLI flag which can be passed to the maintenance tool which gets installed together with the actual LibrePCB application.

Dragon1573, 2023-11-09:
> I'm not sure if this is what you're looking for, but I think there's a `--silentUpdate` CLI flag which can be passed to the maintenance tool which gets installed together with the actual LibrePCB application.

Sorry, I can't get it. Does it mean `.\librepcb-installer-1.0.0-windows-x86.exe --silentUpdate`? There's no response from the console ...

ubruhin, 2023-11-09:
> Sorry, I can't get it. Does it mean .\librepcb-installer-1.0.0-windows-x86.exe --silentUpdate? There's no response from the console ...

No, not the installer exe. After the initial installation there's a file `C:\Program Files (x86)\LibrePCB\librepcb-maintenance.exe` which supports the `--silentUpdate` flag. But there's no console output because it's a GUI tool and GUI tools on Windows can't have console output (at least not with very ugly workarounds).

Note that we're not experts on these installer tools as they are coming from Qt (google QtIFW) and we're not happy with them anyway so I hope some day we can replace them with a better installation tool. See #1256 for more information.

luzpaz, 2024-02-10:
This ticket is blocker for downstream repos like winget (and possibly chocolatey as well).

ubruhin, 2024-03-26:
A new installer will be available with the next release, see https://github.com/LibrePCB/LibrePCB/issues/1256#issuecomment-2015188804. Thus closing this.

closed 2024-03-26 by ubruhin