LibrePCB/LibrePCB #1266: Refactor license selection in new project wizard

BuFran, 2023-11-06:
##### SUMMARY
Current implementation of UX in New Project Wizard reached it's limits and needs little bit rework.

The main issue is the separation of licenses from the code. Currently all licenses with their info needs 
to be hard-coded into the source code. Adding new license as in #1247 needs to modify source code 
and attached files as well and rebuild entire codebase.

Second issue here is that some licenses may need to add additional files to the project to be fully licensed.

Some work has been done in unmerged outdated PR #666, I made some assumptions along that PR.

I propose changing license content from "file" to "folder" inside resources and load all directories 
dynamically as workspace configuration on application start. All files in that directory should be 
copied to the project directory.

The folder should contain metadata about license (metadata should not be copied along other files to the
project dir)

This enables following setups:

* The user can add his own license just by putting files to the [licenses](https://github.com/LibrePCB/LibrePCB/tree/master/share/librepcb/licenses) folder, no code change needed (best for proprietary licenses)
* The license can contain more files when needed (license PDFs for example)
* The link in UX can point to installed directory with correct license text (no web connection required)
* Licensing code can "detect" more actions required situation some way from the license directory
* The installer can have "checkbox" options to install only licenses the user really wants to use in production.
* Users can share and distribute "license packages" without touching librepcb team at all (it is only zipped folder).
* There can be button in UX "install new license" that can unpack zip-packaged license into that directory
* The workspace can have second folder with user's owned license templates
* Each license can have own checks for correct application of license
* There is no need to change file format
* In future, the dropdown can be multicolumn, with filtering to search by criteria (ie view only commercial, shareable etc)

##### SOLUTION
My proposal is to remove a lot of code for filling license dropbox and make it automatically searched from the directory [licenses](https://github.com/LibrePCB/LibrePCB/tree/master/share/librepcb/licenses) where each license would be in separate directory. 

Entire directory content (nonhidden, ie not license metadata) will be copied to the project when license be accepted. 

The directory contains metadata stored as "ini" file with needed information.

My solution is in my work-in-progress branch [ux-license](https://github.com/BuFran/LibrePCB/tree/ux-license)

[image]

(sorry for czech translation in screenshot, I am testing translations along development to see untranslated strings)

I propose changing "info icon" into "dropdown button" with some dynamic context menu assigned to the license to directly access/open the license files.

As it is work-in-progress, there is no tests yet. It is based on PR #1247 that needs to be merged first.

The text "additional actions might be required..." will move to bottom of tooltip text when some autodetection occurs (ie board should contain `{{LICENSE_TEXT}}` expansion placed somewhere via component.

##### OPEN PROBLEMS TO DISCUSS
   
   * [ ] How the metadata should be stored along directory ?
      * accept ini file
      * we should have different format ?
   * [ ] How we should name the keys ? 
      * currently the ini has namespace `LibrePCBLicense` and has following required keys:
         * `Name` as displayed name
         * `Link` as the URL of license on the web. Synonym can be `Url`, maybe it will be better
      * I would like to build `Attributes` namespace where custom variables for expansion in board and schematic (like `{{LICENSE_TEXT}}`) can reside
      * There should be `Check` namespace to enablers of DRC check to apply license correctly
   * [ ] How we should manage translations ? 
      * my proposal is to have keys with translations inside the metadata. Not the best but it will work.
      * The key can be `Name.en_US` for example
   * [ ] How we should manage versions of license, timestamp ?
   * [ ] How we should **document** the license directory format ?
   * [ ] Is the "license package" something we need ? 
   * [ ] Is the tooltip text sufficient or we need to make custom form/display for hint ?
      * I know there will be big refactoring on UI ...

BuFran, 2023-11-08:
Another way how to do is to add another page into wizard "Add license", just after metadata and before initialization, but the order can be any.

[image]

Benefits:
 * The user can be informed about the license parameters and files to be copied in this step. 
 * The user can doubleclick on the license file to directly open it from local filesystem
 * The URL is visible so user can open webpage by doubleclicking
 * The list can have optionally checkboxes that user can select which files he would to copy to project (defaults to all)
 * User can compare license basic features by looking at checkboxes.
The question is "additional actions ..." placement, maybe it would be another feature checkbox to save space ...

ubruhin, 2023-11-10:
Thanks for sharing your thoughts about this topic! I agree 100% the current implementation directly within the dialog's source code is very ugly and requires a better separation of code and license data. 

I just think we need to find a solution which isn't unnecessary complex - in the end, probably most users don't care about the license at all (which is totally fine for personal, non-public projects) and for the rest the licenses suggested in #1247 are sufficient. The only case I see for adding custom licenses are proprietary licenses. The question is whether we should actively support this use-case - using custom licenses is possible anyway by manually adding the corresponding files to a project.

> I propose changing license content from "file" to "folder" inside resources and load all directories
dynamically as workspace configuration on application start. All files in that directory should be
copied to the project directory.

Sounds reasonable :+1:

> * The user can add his own license just by putting files to the [licenses](https://github.com/LibrePCB/LibrePCB/tree/master/share/librepcb/licenses) folder, no code change needed (best for proprietary licenses)
> * Users can share and distribute "license packages" without touching librepcb team at all (it is only zipped folder).
> * There can be button in UX "install new license" that can unpack zip-packaged license into that directory

Note that the `share` folder is not intended to be written by users. With various distribution methods, this directory is therefore not writable by a user (mainly on Linux, but I think on Windows it's also either write-protected or at least considered bad practice to write something into `C:\Program Files (x86)\LibrePCB\`). So this should not be considered as an official way to add custom licenses.

> * The workspace can have second folder with user's owned license templates

This would be much better :+1:

> How we should manage translations ?
>
> * strings are not in ts file, so we can't use transifex with main app
> * my proposal is to have keys with translations inside the metadata. Not the best but it will work.
> * The key can be Name.en_US for example

This is a good question. And from my point of view, it's the biggest drawback of the strict separation of code and data. Having translations inside the metadata makes it difficult to actually contribute translations. If they are checked into this repository (i.e. outside of Transifex), no translations will be added because it's too cumbersome. If they are not checked into this repository, we'd need a way to get the strings from these files pushed to Transifex and fetched back during the deployment. But this is a pain to set up and maintain.

In this regard, having the licenses part of the source code has a big advantage as translations work out of the box.

This makes me thinking about a compromise: Maintaining licenses in the source code, but in a separate file which basically only contains data, not code. Then it would still be a clean separation of code and data, but in a simple way with translations working. Of course with the drawback that licenses cannot be loaded dynamically at runtime - so the big question is whether we need dynamically loaded licenses or not. I'm not sure about that yet :thinking:

> How we should manage versions of license ? 

I don't think versioning is needed. For every newly created project, the latest license file will be added. And for older project's usually it's no problem to stick with an older version of a license.

Of course major license versions (like "GPL v2" vs. "GPL v3") do matter, but they can just be handled as separate, unrelated licenses.

> Another way how to do is to add another page into wizard "Add license", just after metadata and before initialization, but the order can be any.

To be honest, I think this is too overengineered :see_no_evil:  Before implementing this complexity into the tool, I'd rather point to some online resource where the user can learn more about the licenses (could for example be our documentation). This way the license choosing procedure doesn't disrupt users which don't care about licensing, while still providing a helpful guide for those who do care. Actually it would be good to add licensing documentation to https://librepcb.org/docs/ anyway.

closed 2023-12-11 by BuFran