LibrePCB/LibrePCB #1271: Support adding images to schematics

ubruhin, 2023-11-21:
##### SUMMARY

See https://librepcb.discourse.group/t/graphics-import-for-schematics/621.

##### SOLUTION

To be defined in detail. Probably simply allowing a few image formats (which are supported by Qt on all platforms) to be added to a schematic, with the following properties:

- Filename (or maybe implicitly using a UUID)
- Position X/Y
- Scale factor

The image file needs to be copied into the project when added to a schematic. Probably it makes sense to copy it into the schematic's directory to maintain a strong relation to the schematic. Alternatively it might be copied into the `resources` directory but then it is not automatically deleted when deleting the schematic it was added to.

hpirlo, 2024-01-27:
Hello @ubruhin 
I want to contribute to this issue. I think that it is better to add the image to the `resources ` directory. then we can add an option to the software that if a schematic removed, dependent files also be removed.
What is your idea about it?
Can I start coding?

ubruhin, 2024-01-28:
Hi @hpirlo 

> I think that it is better to add the image to the resources directory.

I have now thought about this, and I came to the opposite opinion :see_no_evil: I think it makes more sense to store images in the schematic directories. The implementation will be *way* simpler (and thus less error-prone), and from the user perspective it doesn't matter that much where these images are stored.

> then we can add an option to the software that if a schematic removed, dependent files also be removed.

I think this is unnecessary complexity. Dependent images should always be deleted automatically, because the image is just a part of a schematic and for me as a user it would be clear that deleting a schematic will also delete its contents.

> I want to contribute to this issue.
> Can I start coding?

This is great to hear! However, to be honest I think this is not a good issue to start with. It might sound simple, but this feature requires to touch many aspects of the software: file format, undo/redo, GUI, graphics rendering, testing. In our experience, pull requests adding this kind of feature are very often never reaching a mergeable state and are thus wasted time (both on contributors side and on our side). Also note that due to the file format modification, we can't merge such a feature anyway as long as we're still releasing v1.x releases so this needs to be hold back for v2.x.

So from my point of view, I'd strongly suggest to start with easier issues not affecting the file format. This would also help to start learning our software architecture and contribution workflow etc. before working on more complex features.

hpirlo, 2024-01-28:
@ubruhin Thanks for your great advice.
Now I will try to contribute to easier issues.
I will inform you if I want to start working on any issue.

z3ugma, 2024-12-18:
Can I transform a part of a footprint into a board outline? I suppose via an export and import from a footprint into a board?

ubruhin, 2024-12-18:
You can copy&paste selected footprint elements (e.g. polygons) directly from the footprint editor into the board editor. Or you could create footprints with a polygon on the "board outlines" layer, which will be used as the board outline when you add the footprint to the board.

z3ugma, 2025-05-06:
For anyone else who comes across this, I've been following the suggestion here: https://github.com/LibrePCB/LibrePCB/issues/1271#issuecomment-2551046893 in the comment directly previous to this one. 

I've also had some success making a "board" in the footprint editor, then saving it and looking at the resulting polygons in the `.lp` file from the package. 

```
 (polygon 2ebd698d-829e-4f43-809c-a6f90ea34475 (layer top_legend)
   (width 0.254) (fill false) (grab_area false)
   (vertex (position -1.5 -1.177) (angle 0.0))
   (vertex (position -1.5 -1.5) (angle 0.0))
   (vertex (position 1.5 -1.5) (angle 0.0))
   (vertex (position 1.5 -1.177) (angle 0.0))
  )
```

then, I can paste the polygon into the .lp file for the PCB and change the layer to `board_outline` and re-open the PCB

ubruhin, 2025-05-06:
> I've also had some success making a "board" in the footprint editor, then saving it and looking at the resulting polygons in the .lp file from the package.

Huh, that's way more complicated than what I meant. Actually you should be able to copy the polygon *directly* in the footprint editor (select it & Ctrl+C) and paste it into the board editor. No file editing should be needed.

z3ugma, 2025-05-06:
I can't exactly remember why I needed to do this - I think it's an incompatibility with the board cutouts layer on a footprint vs on a PCB. I could have the slots/holes in the PCB but not in the footprint maybe? so I had to get the arcs drawn correctly over the reference image in the footprint editor, and then copy the s-code, and change the layer from a documentation layer to a board cutouts layer?