LibrePCB/LibrePCB #1279: Workaround to build on macOS from source

ekawahyu, 2023-12-20:
##### VERSION / OS / ENVIRONMENT
macOS Monterey

##### SUMMARY
I followed exactly the steps to build on macOS:
```
brew update && brew install qt5 cmake opencascade
brew unlink qt && brew link --force qt5
git clone --recursive https://github.com/LibrePCB/LibrePCB.git && cd LibrePCB
git submodule update --init --recursive
mkdir build && cd build
cmake ..
make -j8
```
but it failed on `cmake ..` step.

I have added the PATH to Qt5, exported, with no success. See below for the error messages:
```
CMake Error at /opt/homebrew/lib/cmake/Qt5Core/Qt5CoreConfig.cmake:14 (message):
  The imported target "Qt5::Core" references the file

     "/opt/homebrew/.//mkspecs/macx-clang"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/opt/homebrew/lib/cmake/Qt5Core/Qt5CoreConfigExtras.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  /opt/homebrew/lib/cmake/Qt5Core/Qt5CoreConfigExtras.cmake:56 (_qt5_Core_check_file_exists)
  /opt/homebrew/lib/cmake/Qt5Core/Qt5CoreConfig.cmake:227 (include)
  /opt/homebrew/lib/cmake/Qt5Concurrent/Qt5ConcurrentConfig.cmake:90 (find_package)
  /opt/homebrew/lib/cmake/Qt5/Qt5Config.cmake:28 (find_package)
  CMakeLists.txt:115 (find_package)
```
The workaround to successfully build on my system is by explicitly adding the PATH as follows:
```
PATH="$(brew --prefix qt5)/bin:$PATH" cmake ..
```
I hope this helps anyone who might encounter the same issue, but I am wondering what is missing on my setup?

ekawahyu, 2023-12-20:
Unfortunately not, I was hoping that was the case. In fact, I have the PATH within .zshrc and still does not work. Literally, I had to have the PATH in there when calling `cmake`.

I am on Apple M1 by the way, if that really matters?

The good news is that, it builds. But either something is wrong with my homebrew, or  something else going on.

ubruhin, 2023-12-29:
Okay strange, when I built on macOS I didn't have to apply this workaround (also on Apple M1). So I'm not sure if this is something which needs to be added to our build instructions - and even if this behavior is intended by Homebrew, I'd prefer a permanent solution instead of extending `PATH` on every cmake invocation...

ekawahyu, 2023-12-29:
I guess we can just close this ticket, at least there is a record that it may happen for some reason.

ubruhin, 2024-02-13:
For the record, on CI now the same issue suddenly occurred as well :sob: I was able to fix it by prepending `/usr/local/opt/qt@5/bin` directory to `PATH`.

When installing `qt5` with Homebrew, it even prints a message which suggests to run the following:

    echo 'export PATH="/usr/local/opt/qt@5/bin:$PATH"' >> /Users/runner/.bash_profile

To make it more portable, I'd replace it by this one:

    echo 'export PATH="$(brew --prefix qt5)/bin:$PATH"' >> ~/.bash_profile

If someone could verify that this fixes the problem **persistently**, it might make sense to add it to the official build instructions.

nimish, 2024-03-23:
A more comprehensive fix is to add `brew --prefix` to `CMAKE_PREFIX_PATH` and potentially also `$(brew --prefix qt_)` if you want a non-default-linked qt.

This should be done at the level of invoking cmake since cmake really ought not have machine specific stuff as much as possible.