LibrePCB/LibrePCB #1286: 3D object color is applied incorrectly when there are identical objects in the STEP file

ekawahyu, 2024-01-15:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version:
Release: 1.0.0
Operating System: macOS Monterey
```

##### SUMMARY
I created a simple SMD capacitor-like object, with 3 different colors on it, red-yellow-blue. It seems that only two colors appear in the library editor. I verified that the STEP file has 3 different colors with FreeCAD. See the following figure:

[image]

Apparently, this happens when there are two or more identical objects in the file, like those two capacitor leads that I made identical on purpose. However, when one of the lead size is slightly smaller or larger by any number, then all three colors show up just fine. See below:

[image]

##### STEPS TO REPRODUCE
Create new component in library editor and add STEP file that contains at least two identical objects in it but with different color. The second object will have the same color as the first one.

##### EXPECTED RESULTS
Identical object with different color should show up with its own color.

##### ADDITIONAL INFO
It seems that no matter how many objects I add with different colors, if they all are identical, only one color is applied to all of them.

[image]

ubruhin, 2024-01-15:
Hi, could you upload your STEP file here to allow us reproducing the issue with the same model?

ekawahyu, 2024-01-16:
Zip file containing the FreeCAD source file and the the generated STEP:

[Capacitor Test.zip](https://github.com/LibrePCB/LibrePCB/files/13944054/Capacitor.Test.zip)

ubruhin, 2024-01-29:
I started to investigate this issue, and it turned out the problem must be located in `OccModel::minifyStep()` because without the minification your STEP model is rendered correctly. I didn't investigate it further yet so I don't know the exact problem, but at least it's good to know the root cause must be located in our own code.

Actually this would be a good issue for contributors since it doesn't require further knowledge about our code base, so any help on this issue would be very welcome!

Btw, in the mean time there's a workaround available:

1. Load the STEP model into the package editor - it will be rendered wrong.
2. Click *File -> Show in File Manager* to open the package in the file manager.
3. There will be a *.step file in the directory (with a UUID as filename). Replace this file with your original filename, but keep the UUID filename (i.e. replace only its contents).
4. Close and re-open the package in the library manager. The STEP model should now be rendered correctly.

closed 2024-03-20 by ubruhin

Xenia1423, 2024-05-27:
Hello, @ekawahyu ! Can you give me your email address? Or you can contact me at: brilliantstone@yahoo.com. I have some questions about some of your projects. Thank you!

PR #1335 (2024-03-20): Fix possibly wrong colors after STEP minification
Seems that too many STEP nodes were merged, thus now blacklisting all nodes containing the term "REPRESENTATION". Slightly increases the size of the resulting STEP models, but I think there's no way to avoid that...

Also added a unit test to avoid regression.

Fixes #1286