LibrePCB/LibrePCB #1292: Schematic Editor: add function "add waypoint" or "break"

Schorsch99, 2024-01-31:
##### SUMMARY
After moving a component, the wire may be diagonal. I could not find any possibility to add waypoints/bending points so that the wire can be routed orthogonally. A new (orthogonal) wire needs to be drawn and the diagonal wire deleted.

##### SOLUTION
Add function "add waypoint" (called "break" in KiCad) or similar in a wire context menu. Also add the opposite: "remove waypoint" and "remove all waypoints" to help clean up some misrouted wires with too many waypoints.

[image]

ubruhin, 2024-01-31:
Full ACK, this should be implemented :+1:

moshee, 2024-03-03:
If a component being dragged will cause a single wire segment to need to gain another axis of travel, I propose automatically breaking the wire near the middle instead of (or in addition to) providing a manual function to do so. 99% of the time this is what I want anyways.

IMO wires should be drawn and dragged orthogonal only by default unless strongly specified (e.g. don't present diagonal wire options in the cycling list of directions and only allow free-draw with a modifier key). There are typically only a small number of specific instances where diagonal lines in schematics are desired.

RDamman, 2024-07-13:
I hope it can be combined with the option to merge 2 or more segments in 1 line.

NilsKr, 2026-03-25:
In selection mode, When clicking a wire to drag it, why not use a keyboard modifier (e.g. Alt/Option) to split the wire segment at that position and select the new waypoint for dragging, making sure it snaps to the grid unless snapping is turned off?