LibrePCB/LibrePCB #1293: Schematic Editor: add function to disconnect/reconnect wire

Schorsch99, 2024-01-31:
##### SUMMARY
Once a wire has been connected to a pin, it can not be disconnected any more. If the pin was the wrong one, a wire segment must be deleted and a new wire drawn to the alternative pin.

##### SOLUTION
Provide the possibility to "pull" the wire from the component (maybe with CTRL+Mouseclick to avoid accidental disconnection) and to rewire it to another pin or component.
I know that LibrePCB behaves the same way as KiCad in this respect, but I find the possibility to easily reconnect wires (similar to the connectors in Visio) quite appealing.

ubruhin, 2024-01-31:
I'm not sure if I understand correctly - are you talking about disconnecting wires in the wire tool, i.e. to immediately undo a connection made by mistake, or do you like to disconnect wires in general, i.e. in the "select" tool? For the former, this sounds similar that what have been requested in #606 and #1028, just for the schematic editor so I think it should be implemented similarly (e.g. same keyboard shortcut).

Schorsch99, 2024-02-01:
What I meant was to disconnect the wire with the select tool. The procedure could be something like this:
- click on a wire or segment to select it > the end points of the segment are highlighted
- now you can click and drag on the endpoints to move them to another component or pin

RDamman, 2024-07-13:
And in the same way you it should be possible to connect a net to a component.

Schorsch99, 2025-07-02:
Nothing seems to be happening wrt basic schematic editing features. Are any editor improvements/additional features planned for V2?