LibrePCB/LibrePCB #1297: Library editor: Increase pin name length

sigmate, 2024-02-07:
I'm creating a small library for the Daisy Seed platform, using the following pinout diagram: 

[image]

https://github.com/electro-smith/DaisyWiki/blob/66a47a96f7981abd0c289a1a28b4e07edbdade8c/resources/Daisy_Seed_Pinout.png

I tried to label pin 35 as `A11/D28/SAI2SCK/ADC11` but couldn't because the name field maximum length is 20 characters.

https://github.com/LibrePCB/LibrePCB/blob/aa3119fd6a92667bd117b15acb37329370e6bc3d/libs/librepcb/editor/library/sym/symbolpinpropertiesdialog.ui#L32

Could that be increased? It's quite common that development boards have a lot of multi-purposed pins. Having their definition included in the pin name seems useful, I suppose?

ubruhin, 2024-02-07:
Thanks for reporting @sigmate, the limit of 20 characters was somehow a mistake in the GUI since actually we support up to 32 characters. I fixed it in #1298 so the next release will allow entering up to 32 characters.

In the mean time, if you already like to use longer pin names, feel free to open your symbol in a text editor (*File -> Show in File Manager* from the symbol editor, then open `symbol.lp` with a text editor). Just adjust the pin name there and save the file.

sigmate, 2024-02-07:
Excellent! Thanks a lot!

closed 2024-02-07 by sigmate