LibrePCB/LibrePCB #1299: Add Repology badge to README showing package status in downstream package repos

luzpaz, 2024-02-07:
https://repology.org displays what version of a specific software is available in the packaging ecosystem. Can be helpful to track if any downstream repos need updating.

### Large badge 

[image]

### Small badge

[image]

ubruhin, 2024-02-07:
I worry a little bit whether the large badge would be too "noisy" in the README or not :thinking: But it might be useful indeed, so I think I'd accept a pull request. I'd probably go with the large badge then as it provides the overview directly.

luzpaz, 2024-02-07:
Where would you want the badge placed specifically in the README ?

ubruhin, 2024-02-07:
I'd put it just below the current badges on a new line, separated the same was as the current two badge groups.

luzpaz, 2024-02-08:
> I worry a little bit whether the large badge would be too "noisy" in the README or not 

Note: If this becomes too noisy, you can always hide the badge within a collapsed div like so:

<details>
<summary>Expand to view downstream package repo status</summary>

[image]

</details>

PR #1300 (2024-02-08): Add Repology badge to README.md
Closes #1299

Preview:

[image]
commit 6911414c (2024-02-08): Add Repology badge to README.md

Closes #1299
closed 2024-02-08 by ubruhin