LibrePCB/LibrePCB #1315: LibrePCB-1.0.0 does not support OpenCascade 7.8.0

thierry-FreeBSD, 2024-02-26:
##### VERSION / OS / ENVIRONMENT
Tested on FreeBSD, but it will give the same result on any OS.

```
LibrePCB Version: 1.0.0

##### SUMMARY
Compilation fails with the message:
```
ld: error: unable to find library -lTKXDESTEP
c++: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
*** Error code 1
```

##### STEPS TO REPRODUCE
- install OpenCascade 7.8.0
- try to build LibrePCB

##### EXPECTED RESULTS
Build succeeded!

##### ACTUAL RESULTS
See the reported error.

##### ADDITIONAL INFORMATION
N/A

ubruhin, 2024-02-26:
Arghh seems they suddenly reorganized library modules :sob: If I understand correctly, `TKXDESTEP` was renamed to `TKDESTEP` but maybe that's not the only change...

See https://dev.opencascade.org/doc/overview/html/occt__upgrade.html#upgrade_occt780

ubruhin, 2024-03-10:
@thierry-FreeBSD This should be fixed in the next release. If you already want to try it with LibrePCB 1.0.0, just apply the patch https://github.com/LibrePCB/LibrePCB/pull/1329.

(not yet verified if this also works at runtime, will be tested later)

ubruhin, 2024-03-11:
I have now tested with OCCT 7.8.0 and although it links without errors, LibrePCB crashes when opening a STEP file :sob: Seems this is a bug in OCCT 7.8.0: https://tracker.dev.opencascade.org/view.php?id=33631

I think I'll "blacklist" OCCT 7.8.0 in CMake (throwing an error if this version is used) and we'll need to wait for the next release.

PR #1329 (2024-03-13): CMake: Fix linking with OCCT 7.8.x
Verified locally on Linux that with this change LibrePCB builds & links successfully with OCCT 7.8.0. Later this will be verified on CI (including Windows) as well when newer Docker images are available (https://github.com/LibrePCB/docker-librepcb-dev/pull/43).

Fixes #1315
closed 2024-03-13 by ubruhin