LibrePCB/LibrePCB #1321: Library: "Copy to other library" unexpectedly retains same UUID in pasted symbol

moshee, 2024-03-03:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 1.0.0
Git Revision:     9edb6ede3
Build Date:       2023-09-24T08:07:32
Build Author:     LibrePCB CI
Qt Version:       5.15.2 (built against 5.15.2)
CPU Architecture: x86_64
Operating System: Windows 10 Version 2009
Platform Plugin:  windows
TLS Library:      OpenSSL 1.1.1v  1 Aug 2023
OCC Library:      OCCT 7.7.2
```

##### SUMMARY
Using the "Copy to other library" keeps the UUIDs when copying a symbol, making it not very universally unique any more and causing things referencing the original symbol to break.

##### STEPS TO REPRODUCE
1. Use "Copy to Other Library" on "MOSFET N-Channel" from the base lib
2. Make edits to the pasted symbol and create a component using it

##### EXPECTED RESULTS
Only the new component uses the pasted symbol

##### ACTUAL RESULTS
* The original symbol is untouched (expected):
[image]
* Existing components in the base lib are now using the pasted symbol and broke as a result (unexpected):
[image]
* The pasted symbol also shares UUIDs of pins, polygons, etc.
* There is also this warning which seems very strange considering copying should expressly not be a potentially destructive action:
[image]

##### ADDITIONAL INFORMATION

New symbol: [symbol.lp.txt](https://github.com/LibrePCB/LibrePCB/files/14471875/symbol.lp.txt)

ubruhin, 2024-03-03:
Keeping UUIDs is intended because the main use-case of this feature is to copy elements from a read-only library (i.e. libraries installed through the library manager) into a local library to modify them. In this case, it is desired that all other elements will reference the own copy instead of the original element.

There are also two ways to copy an element while automatically assigning new UUID (effectively making it a *new* element):

- Right click on an element in the library manager and choosing "Duplicate". However, this duplicates the element within a library, not across different libraries.
- In the library editor, choose "File -> New Library Element" and then select "Copy from existing element". This allows to copy an element from any library into the currently opened library.

Anyway, I agree it is not very intuitive and probably we should make the text in the message box much more descriptive.

moshee, 2024-03-03:
Some time after posting this issue last night I wondered if that might be the case. I think it would help to improve the wording to make the dependency resolution method more clear in context.

Suggestions for the existing menu option that I used:

* *Edit as copy in local library*
* *Copy and override in local library*

The warning message can then explain in more detail how the name resolution works.

I think it would be nice to keep the existing option to do what I originally wanted, but this might be confusing for people who actually use this feature as implemented regularly. Something like *Duplicate in local library* might be better as it follows the existing nomenclature of *Duplicate*.

PR #1343 (2024-03-28): Library editor: Refactor "copy to other library"
Remove the "Copy to other library" context menu item because it was misleading (users could think this feature allows to create a new library element which will cause serious trouble, see #1321). Now only the "Move to other library" option is available in the context menu, but its confirmation dialog contains a checkbox to keep the elements in the current library (effectively making a copy):

[image]

It might still not be a perfect solution but I didn't find a better one and I think this should now be much less "dangerous".

Note that this chec [... cut]
closed 2024-03-28 by ubruhin