LibrePCB/LibrePCB #1328: Cross-probe between schematic and board

ubruhin, 2024-03-09:
##### SUMMARY

For revieweing and debugging a PCB it's very useful to cross-probe nets betwen schematic and board so this is a basic feature of every EDA tool and should be added to LibrePCB too. Just opening this issue to keep it in mind.

##### SOLUTION

Tbd. Nice would be to cross-probe nets even without invoking a tool, just by selecting something (e.g. selecting a schematic wire could automatically highlight that net in the board editor.

EdizonTN, 2024-03-09:
**Even better solution:**
Nice would be to cross-probe nets even without invoking a tool, just by **mouse move over** the wire/label/track/via/pin... (simply, all elements which are electrically connected to net) in both direction (from Pcb to Sch and opposite way also).

moshee, 2024-03-10:
I can think of a few levels of interaction that can be useful:

* Cross probe highlight (e.g. on hover)
* Cross probe select (e.g. on click)
* Cross probe select and navigate to/zoom (e.g. on modifier click)

Some other EDA tools only support the always-on "click to select and navigate to/zoom" level, which is extremely annoying when you're e.g. trying to do a bit of layout and different schematic sheets keep popping up.

Support all relevant objects (including components) too, not just nets.

xotho, 2024-03-18:
Include also the **single component cross-probe**.

Cross-probing have to be BIDIRECTIONAL. (From SCH to PCB and From PCB to SCH)

xotho, 2026-04-08:
> Cross-probing between schematic and 2D board editor is implemented in
> #1748 <https://github.com/LibrePCB/LibrePCB/pull/1748>. For the 3D view I
> opened a separate issue #1750
> <https://github.com/LibrePCB/LibrePCB/issues/1750>. And for "zoom to
> cross-probed objects" I opened #1751

I think this can be really useful!

X!