LibrePCB/LibrePCB #1341: Shematic Editor: different colos per wire

SirHerpDerp, 2024-03-24:
##### SUMMARY

For me it would be nice if there is a possibility to change the color of every wire in the shematic editor for better orientation. I'd really like to give data, vcc and gnd wires different colors.
Haven't found any Issues or PR for this (or the feature itself).

##### SOLUTION

Since I'm new to all of this I only used EasyEDA right now. There you can click on a wire and see the color in the option pane.

[image]

When clicking at the color option you will get a color picker.

[image]

ps. thank you (and all contributers) very much for creating LibrePCB, it looks very good to me (especially the easy library manager). Have a nice weekend.

ubruhin, 2024-03-24:
Hmm somehow it sounds a bit unusual to me since a schematic editor is not a graphics drawing tool :see_no_evil: But it might indeed be useful, I'm not sure. Though maybe I'd implement a color property rather on a per-net basis instead of per-line...

dbrgn, 2024-03-24:
A per-net color in the schematic might actually be useful, yes!

xotho, 2024-03-25:
Color can identify unrouted nets maybe or even "special nets" (unspecified necessity but who apply a color well know the reason).

SirHerpDerp, 2024-03-25:
> Hmm somehow it sounds a bit unusual to me since a schematic editor is not a graphics drawing tool 🙈 But it might indeed be useful, I'm not sure. Though maybe I'd implement a color property rather on a per-net basis instead of per-line...

Haha, yes I can completely understand you point of view and may the inventor of shematics turn her/himself in the grave when hearing that people are colorizing shematics.

A per-net color would be nicem would not know a use case for a per-line color right now (but who am I to judge, just an amateur who has 'drawn' two shematics).
I just can speak from my point of view but this was a feature at EasyEDA I learned to love very fast (and which prevents me from switching to LibrePCB somewhow :P). Picked up a very old project recently to improve it. It uses an infrared receiver and you can imagine I do not remember the pinout anymore. With the colored wires I can see it within a blink of the eye (red wire = vcc, black wire = gnd, other wire = data) without following the wiring or looking up the docs.

ubruhin, 2024-03-25:
> [...] and which prevents me from switching to LibrePCB somehow :P

Huh, that would be really bad to decide against LibrePCB because of such a trivial feature :see_no_evil: :sweat_smile: 

> With the colored wires I can see it within a blink of the eye (red wire = vcc, black wire = gnd, other wire = data) without following the wiring or looking up the docs.

Well, actually I'd recommend to use [net labels](https://librepcb.org/docs/quickstart/create-project/schematics/#add-net-labels) for that purpose. It makes schematics so much more readable.

With colors, I see two problems:
1. In printed form with only grayscale printer, colors are not visible so you loose the information "red wire = vcc, black wire = gnd". With netlabels, colors don't hold important information so a (printed) grayscale schematic is as expressive as a colored schematic.
2. When nets may have colors, the color information (obviously) needs to be stored in projects. But all other colors are defined by their layer, which are configured in workspace settings. So every LibrePCB user can configure their colors as they like, and all projects will use these colors. However, when a project overrides the colors of particular objects in a schematic, conflicts can occur. For example if a user configures the "symbol outline" layer to have red color nets set to red color (e.g. VCC) will have the same color as symbol outlines, which is a mess and/or misleading. Another problem is compatibility with light/dark color schemes - net colors won't be theme-agnostic so their color might be completely unreadable on some user's systems. This is really something which makes me thinking colored nets are not a good feature.

xotho, 2024-03-26:
Net Labels Are perfect in general, just do not show you directly the info, interpret is a must.

I'd consider colors just in the SCHEMATIC (while on the screen), obviously different colors have to be unified in case of printing (in this case labels are perfect). My idea should be to have just a small set of different colors paired by it's high-lighted ones for cross-probing (sch and brd).

but this is just an idea to identify at least a subclass of lines. (the same can be applied to BUS probably)

I like anyway FREEPCB even if I use Cadence etc.

PetoJanis, 2024-03-27:
Hello @ubruhin,

Yes, I vote for possibility of change color for selected net (SCH and PCB). Each good feature which is as standard in the others EDA system, is expecting in the LibrePCB. Now or in the future (this is your decision).

My experience in the schematic editor:
I use colored (and thicker !!) net wire - mostly I highlight nets for higher power. **Schematic is better and quicker readable**.
In the PCB editor: Colored net of unrouted airwires is helpfull for better routing (and possibility hide selected airwires also).

**How to solve ?**
Yes, on the black&white printers can be problem. But this is question for all other schematic object - are in the various color. Solution is: in print dialog add switch for B&W printing - Ouu, you did it - problem solved :-)

Chaos between color scheme and direct set of the color for some net?
Add a checkbox into Workspace configuration -> Theme Tab: "Override theme color with local color setting". If it checked, you draw net wires with color parameter as user set. Otherwise draw it with settings from workspace (Theme). And this is it.