LibrePCB/LibrePCB #1356: PCB Terms unifying - Solder resist vs Stop mask vs Solder Mask

EdizonTN, 2024-04-01:
##### SUMMARY
Please check (LibrePCB v.1.0.0):
In the Board Setup dialog - Tab: General, you use term "Solder resist", but in DRC Settings Tab, you use term "Stop mask".

These two terms are used also in the DRC messages (I found it during translation) many times.
Will be good, if terms are minimalized to one only.
I think, "Stop mask" is equal to "Solder resist".

##### SOLUTION
My proposal: use "Solder mask" instead. It's more logical todays than stop mask....

ubruhin, 2024-04-02:
Actually I cleaned up this terminology some time ago because "stop mask" and "solder mask" were mixed. But the current terminology is intended, because IMO "solder resist" and "stop mask" are not the same thing:

- Solder resist means the **material** on the PCB, i.e. the areas which are *covered*.
- Stop mask means the areas where **no solder resist** is added, i.e. areas which are *exposed*.

I understand if it is a bit confusing and not obvious, but I don't know how it could be improved. I don't like to use the same term for both things because IMHO it would not be correct.

ubruhin, 2024-04-24:
Considering this as resolved as there were no objections.

closed 2024-04-24 by ubruhin