LibrePCB/LibrePCB #1360: 3D view.

xotho, 2024-04-03:
Since the application can export STEP files, can it be possible to implement a set of primitive functions to move , align, rotate PCBs ?

The question is :
if i have more pcb's circuit in the same BOARD, I'd like to place those in the space in the right way with the right distances with the right angles  .... exporting maybe just a final step that gives exactly the position and dimension of the placed pcs's.

This would let you treat the electronic parts as a single assembly, mechanically it can help a lot in many circumstances.

X!

ubruhin, 2024-04-03:
Hmm theoretically this might be possible, but to be honest I think the placement of models is the job of the MCAD. As far as I remember, usually you only need to define placement of models in an assembly once and it stays intact even when updating one of these models.

Thus I doubt such a feature is worth the effort and the additional complexity of LibrePCB. Complexity always comes with more complicated/polluted user interface and risk of bugs.

xotho, 2024-04-04:
Understand your point of view.

So, I see an alternative also. **Export singularly the boards if more than 1 present** including the full assembly as is. 

So with 1 board, 1 export; with n.boards (n.boards+1) exports....

take care,
X!

ubruhin, 2024-10-17:
I'll close this issue as I still think transform is the responsibility of the MCAD, not LibrePCB (and #1364 has been fixed). Let me know if there are really good reasons to implement it in LibrePCB.

closed 2024-10-17 by ubruhin